OpenBoardView/OpenBoardView #2: Add compatibility with "Honhan BoardView" files (.bdv)

stray77, 2016-05-23:
https://www.dropbox.com/s/slfn9lfmfnyzjzf/BDV-BoardView.zip?dl=0
There's a copy of Honhan's Boardview.exe and a sample .bdv file (compal_la-5911p_r1.0_boardview.bdv).  I hope you're interested in adding compatibility with this file type as well.
Great job and thanks for your efforts.

chloridite, 2016-05-24:
It seems Dropbox has disabled that download link.

stray77, 2016-05-24:
[BDV-BoardView.zip](https://github.com/chloridite/OpenBoardView/files/280025/BDV-BoardView.zip)

maybe this will work? dropbox got too much traffic. lol you're too popular.

chloridite, 2016-05-24:
Thanks, got it decrypted -- will work on integrating the format later.

[compal_la-5911p_r1.0_boardview.bdv.txt](https://github.com/chloridite/OpenBoardView/files/280100/compal_la-5911p_r1.0_boardview.bdv.txt)

inflex, 2016-06-25:
@chloridite do you have the decoding for these two formats added yet/soon?  Thanks for doing them :+1:

piernov, 2016-06-26:
See here: https://github.com/piernov/OpenBoardView/commits/bdv-support

chloridite, 2016-07-05:
```go
package main

import (
	"io/ioutil"
	"os"
)

func main() {
	buf, err := ioutil.ReadFile(os.Args[1])
	if err != nil {
		panic(err)
	}
	line := 1
	for i := 0; i < len(buf); i++ {
		b := buf[i]
		if b == 0xa {
			line++
			continue
		}
		if b == 0xd && i < len(buf)-1 && buf[i+1] == 0xa {
			line++
			i++
			continue
		}
		l := int8(line % 0x7f)
		if true || b <= 0x7f {
			b = byte(0x7f - int8(b) + l + 0x20)
		} else {
			b = byte(0x7f - int8(b) + 0x20 - l)
		}
		buf[i] = b
	}
	os.Stdout.Write(buf)
}
```

chloridite, 2016-07-05:
This is my decoder from reading the disassembled .exe -- might lead to a slightly better-looking `decode_bdv`function.

inflex, 2016-11-02:
Are we okay to close this now?  I believe this is fully supported as far as available test files have show  as of R7.3 ?

piernov, 2016-11-02:
I think it is too.

closed 2016-11-02 by MattSturgeon