OpenBoardView/OpenBoardView #3: Remove unnecessary null pointer checks

elfring, 2016-05-23:
[Extra null pointer checks are not needed](http://pubs.opengroup.org/onlinepubs/9699919799/functions/free.html "Do I need to check for NULL before free p?") in [the destructor for the class "BRDFile"](https://github.com/chloridite/OpenBoardView/blob/398d812959992a347525eb57f7c378a27f3fe8f2/BRDFile.h#L46 "Update candidate").

closed 2016-05-24 by chloridite

elfring, 2016-05-24:
[Another update candidate](http://dietmar-kuehl.de/mirror/c++-faq/freestore-mgmt.html#faq-16.8 "Do I need to check for NULL before delete p?") was [left over](https://github.com/chloridite/OpenBoardView/blob/98a30693f847a95e48bae4c897b3cd63c13128bd/BoardView.cpp#L13 "More unnecessary checks?").

elfring, 2016-05-24:
Thanks for [another small source code improvement](https://github.com/chloridite/OpenBoardView/commit/92732ad7161637db28b944f8f3347250cde4225b "Deletion of two unnecessary checks").

pafmaf, 2016-05-29:
@elfring 

In `BoardView::SetFile(...)`, if you don't check before the `delete`, how do you know `m_file` is a valid pointer, i.e. not already deleted?

If there was `BoardView::CloseFile()`, that would presumably get rid of `m_file` by deleting and setting it to null. So if you don't check for null next time, it gets deleted twice.

elfring, 2016-05-29:
Should [the involved resource release functions usually check for a passed null pointer](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for NULL before delete p?") by themselves?