OpenBoardView/OpenBoardView #6: missing things

psycodeoday, 2016-05-25:
thank You for this app! You're awsome

three things are missing

1.ability to type few (more than 1) componets to search box.
2 ability to clear marked nets/componets (by key or button)
3.ability to see all components names on the board it the same time.
4.ability to restore to default zoom and position (center out or something like that)

propably the better idea will be to use right mouse to move board, because using left mouse for move can accidently turn off/clear our component/net marks.

slayer200230, 2016-05-25:
also just wanted to add, the zoom function, for people on a laptop, or without a middle mouse key, may be nice to use +,- to zoom in and out, or allow keybindings to be set up by the user.
other than that, great start, much less problems than other brd view software

pafmaf, 2016-06-19:
I can image these things would be very helpful to a lot of people. 

@chloridite It'd be cool if we could get any info on when you possibly find some time to review the PRs by @charliebruce especially. Thanks.

piernov, 2016-10-30:
1, 2, 4 is done as of R7.3. Not sure about 3 and the other suggestions. @inflex do you have more info on this?

inflex, 2016-10-30:
#3 is sort of implemented as you hover over each part it'll provide the part name as a tool-tip following the mouse.  

We're inhibited a little bit here because of the lack of variable size fonts, which prevents us from nicely packing the part name within the bounding-box of the part.

When ImGui provides us with variable fonts we'll be able to fully implement.

romanstingler, 2020-08-27:
@piernov can we close this issue? imho if there are further suggestions new issues should be created