OpenBoardView/OpenBoardView #7: Not an issue, Just  a question.

drewrox2009, 2016-05-26:
I love what you have done with the software! I have a question that you may or may not be able to answer. If some more values for components were compiled, 1. would there be a simple solution to modify the .brd files to hold this information, 2. Would it be possible to have your software display this information with selected components.

i.e. the voltage that should be measured at test points, caps, pins in a certain power state. and possibly resistance or package type/size of certain components. 

I have very limited programing knowledge, outside of what i did years ago. But i was seriously considering putting some effort into building a solid app that would give you a bit more information without having to jump back and forth to the schematic. by no means a replacement for actually having a schematic as well. 

again, we all appreciate the work that you have done here!

piernov, 2016-06-03:
I think it is what @pafmaf wants to achieve with the annotations.json file. He exposes some pretty nice ideas on #11 so you should leave a comment there.

pafmaf, 2016-06-04:
> 1. would there be a simple solution to modify the .brd files to hold this information

Not quite modifying `.brd` I think, but yes, as piernov said, how to store that (and other extra info) is one issue (11).

> resistance or package type/size of certain components.
> the voltage that should be measured at test points, caps, pins in a certain power state.

What do you exactly mean by 'power state', things like charging / not charging / etc.?

The way you describe it, there's _more than one annotation per component or pin, having a distinct category._

There are values like pad size or component size that are useful for rendering, so I'd separate that from other 'user annotations'.

```
<...>
// parts / components
"<unique_part_id>" : {
	{
		// useful for rendering
		"package" : "really_big",
		"type" : "<resistor>|<cap>|<ic>|<connector>",
		"<field>" : "<value>",
		
		// user annotations
		"annotations": {
			"value" : "10k",
			"note" : "not critical",
			"<annotation_group>" : "<raw_string>"
		}
	}
},

// pins / testpads / vias
"<unique_pin_id>" : {
	{
		// useful for rendering
		"diameter" : "0.3mm",
		"<field>" : "<value>",
		
		// user annotations
		"annotations": {
			"value_charging" : "5.3V",
			"value_battery" : "0.8V",
			"<annotation_group>" : "<raw_string>"
		}
	}
}
<...>
```

:thought_balloon: Thinking about it now, it might even be a good idea to have 2 seperate files for it, ie. `annotations` and `board_extra` or something. Since annotations may be use-case specific, while components and pad sizes are simply data that could've been in the original board file in the first place.

That way we could group annotations by their keys and you could then dis- or enable those annotation groups, e.g. `value_charging` via a checkbox/listbox or something in the UI,

```
Annotation Groups
=================
[X] value_charging
[ ] value_battery
[ ] note
```

without structuring the annotation source too much. So _"states"_ or "groups of annotations" are always user defined.

stray77, 2016-06-04:
Here's a nice description of Power States S0 through S6.
http://arstechnica.com/civis/viewtopic.php?f=8&t=354552

piernov, 2016-10-30:
Annotations feature from @inflex should do what you want. Available in R7.3.

closed 2016-10-30 by piernov