OpenBoardView/OpenBoardView #8: Suggestions.

dirday, 2016-05-26:
If you can add drag and drop and board title in title bar that would be very helpful.

drewrox2009, 2016-05-26:
i would absolutely love this!

inflex, 2016-06-20:
I've added board-name in titlebar as well as file history, but at the moment it's on linux only and I haven't even git'ified my version, but I have the code up and running :)

While I'm not sure how to add 'drag and drop' I am going to be adding the ability to load when invoked (via command line) so that might facilitate the drag-drop ability, or at the very least the ability to load on file association.

[image]

charliebruce, 2016-06-24:
https://github.com/mwh/dragon might help with the drag-and-drop functionality on Linux. 

Awesome to see a native implementation working (you're using OpenGL?) - would be nice to see your code on Github when you're done so that we can get a single codebase which natively supports all platforms. The OpenGL renderer might also help with compatibility, eg #22

inflex, 2016-06-24:
@charliebruce I've got my git fork repo going;  (it's the SDL/OpenGL/ImGui codebase)

git clone -b inflex-ui-features https://github.com/inflex/OpenBoardView.git

( https://github.com/inflex/OpenBoardView/tree/inflex-ui-features )

charliebruce, 2016-06-24:
Ah, hadn't seen that you had uploaded that. Your changes are looking great, I especially love the rotated bounding boxes. Have you tested on Windows at all, or is this Linux-specific? I can help bring back to Windows if you haven't already done so (settling on OpenGL for the renderer on all platforms seems sensible, reducing amount of platform-specific stuff can only be a good idea).

inflex, 2016-06-24:
@charliebruce right now it's just being hashed out on my linux workstation.  I was somewhat waiting till I got to the end of this run of "inspiration" before I really thought about pushing things back to the windows version (I don't have any experience with windows builds).

Hoping within a few days I'll have settled it down.

Even the  indenting is getting quite scrappy now :-(

inflex, 2016-06-24:
@charliebruce just fixed the MBB algorithm, so now finally I can rest for tonight (2am here).  Drag-drop works as well, at least when I fired up Nautilus and dragged a BRD file in to it it worked.

I noticed I have no problem running multiple concurrent instances of OVB on this system either.

MattSturgeon, 2016-08-17:
Implimented in 363213d and 523b6d3

closed 2016-08-17 by MattSturgeon