OpenBoardView/OpenBoardView #9: Cross platform support (i.e. Linux)

MattSturgeon, 2016-05-28:
It would be great to be able to make this cross platform (especially since `imagui` is platform agnostic).

I believe this would involve two (fairly large) changes; moving from DirectX to OpenGL and moving from .NET to a cross-platform framework like QT.

It's always a shame to have open software restricted to windows

pbhj, 2016-05-29:
FWIW I tested with some Macbook .brd files I found and it seems to work well under WINE (1:1.6.2-0ubuntu1, on Kubuntu 15.10).

MattSturgeon, 2016-05-30:
@pbhj Thanks for the tip. Full disclosure: I'm not actually using this software, but I saw it featured on [youtube](https://www.youtube.com/watch?v=gPPleknyEEw) and was disappointed to find windows only software based on a lib that has put so much effort into supporting multiple platforms!

From further inspection it looks like this is based on imgui's [directx9 demo](/ocornut/imgui/tree/master/examples/directx9_example), so it may not be too late to transplant it into the [SDL/openGL boilerplate](/ocornut/imgui/tree/master/examples/sdl_opengl3_example), the bigger challenge would be abstracting or removing all the .NET stuff and using cross platform filesystem (etc) APIs.

piernov, 2016-05-30:
Somewhat done: https://github.com/piernov/OpenBoardView/commit/a5c53970b723e432ea1b9e52d830277434ad21db
(based on https://github.com/pafmaf/OpenBoardView/tree/slight-refac )
Now we have to make it build again on Windows, and add support for loading textures with the OpenGL backend. But i've no idea about the latter, no experience in OpenGL.

chloridite, 2016-05-30:
The reason I say to use Wine is because I don't have a proper build environment for OSX, and would have to spend time putting together Makefiles for OSX and for Linux. I'd also have to figure out how to replace `GetOpenFileName`, and I would have to either do ELF/Mach-O resource embedding or loose files for the two embedded assets the program depends on.  Transplanting the example ImGui code for other platforms is the easy part.

As an aside, there's no ".NET stuff" here, just Win32 stuff.

MattSturgeon, 2016-05-30:
@piernov Very impressive.

Once I'd remembered how to `git submodule update --init --recursive` it worked like a dream, with the exception of textures on the pins as you pointed out (and a small bug I've fixed over in [PR #1](/piernov/OpenBoardView/pull/1))

It might make sense for both the dx9 and opengl engines to make use of SDL, in theory this could allow OBV to use the SDL input events for (hopefully better) keyboard shortcuts - though I don't have any experience in either imgui or SDL, I get the feeling there may be some nice-to-haves tucked away in SDL.

@chloridite Thanks for the response.

> I don't have a proper build environment for OSX, and would have to spend time putting together Makefiles for OSX and for Linux.

Yes, the build environment is a bit of a stumbling block, but @piernov has done a good job of setting up the CMake makefiles for linux, so all you really need now is [VirtualBox](https://www.virtualbox.org/wiki/Downloads) and the OBV build dependencies.

> and I would have to either do ELF/Mach-O resource embedding or loose files for the two embedded assets the program depends on

I assume these are bitmap assets? If so I'm not sure I see a problem with exporting bitmap files into the assets directory. If they are libs, why not the libs directory?

> As an aside, there's no ".NET stuff" here, just Win32 stuff.

My mistake! You may be able to tell I'm unfamiliar with all this Windows stuff :)

-------------------

If on the other hand you simply don't want to support platforms you don't use (perfectly understandable) many other projects will use a teamwork approach: one person handles windows and another linux.

piernov, 2016-05-31:
Works on OSX too(with a few fixes on my osx-support branch) but I'm still trying to figure out how CMake/CPack works to make an app bundle and a DMG. Also it uses GTK3 for the open file dialog instead of the native OSX framework (I might try to implement that later). For now you have to build it the same way as for Linux, and install with homebrew all the dependencies needed (pkgconfig, gtk+3, sdl2 at least).
Btw, I used pafmaf's branch just because the code organization felt better.
Lastly, I know my branch is not suitable for a PR. If you have any advice for making commits suitable for a PR, please notice me.

MattSturgeon, 2016-05-31:
> Btw, I used @pafmaf's branch just because the code organization felt better.

From the looks of pafmaf@9a66e98f4dcf9637be4f7fefa395af6248dc3d1d it looks like he's planning to send a PR soon.

> I'm still trying to figure out how CMake/CPack works to make an app bundle and a DMG

[These wiki docs](https://cmake.org/Wiki/CMake:CPackPackageGenerators) are the best info I can find on CPack package generators. Looks like it _should_ be fairly trivial to support Windows NSIS installers, OSX archives and .deb and .rpm.

Apparently most of the docs are [built in to CPack](https://cmake.org/Wiki/CMake_builtin_documentation_handling)

pafmaf, 2016-05-31:
> From the looks of pafmaf@9a66e98 it looks like he's planning to send a PR soon.

Soon-ish, since I don't find much time. That branch is WIP. Originally I just wanted to have an abstract `Board` class and a listing of components but things escalated a bit. (I'll clean up the history before PRing.)

Working with Windows/VS is a pain, so next up is merging @piernov 's changes, getting that to compile with MinGW on my machine and finding some way of having a fully abstract `Board`. Then it's good enough and usable for me _personally_.

piernov, 2016-05-31:
Updated my linux-support/osx-support branches. Now it displays the texture for the pads, and OS X build doesn't depend on GTK3 since it now uses Cocoa for the open file dialog. Also fixed VS project files, but I couldn't get CMake to build the VS files, and I still have no idea on how to include the assets with the app bundle on OS X.

MattSturgeon, 2016-05-31:
@piernov Good progress!

I've been trying to look into CMake and CPack. I've made _some_, since I don't have a Mac I've not tried building a mac bundle yet, but it looks like the Cmake files need a bit of a refacter, and the build.sh script is becoming more and more redundant.

In [this branch](/MattSturgeon/OpenBoardView/commits/cmake-experiment) I've been running
```
mkdir build && cd build
cmake -D CMAKE_BUILD_TYPE=DEBUG ..
make
cpack --config ./CPackConfig.cmake -G DEB
cpack --config ./CPackConfig.cmake -G ZIP
```

It's very WIP as I'm new to CMake, but though I may as well share in case anything helps.

Current known issues:

- Does not link correctly (`error while loading shared libraries: libimgui.so: cannot open shared object file: No such file or directory`)
- Only tested on .deb and .zip so far
- Last commit was a "going to dinner" type so may be broken

inflex, 2016-06-12:
Jumping in because I want to see this on linux (ubuntu 16.04 as of current).   Downloaded the slight-refac git branch but as soon as I try to run cmake I just end up with  "...does not appear to contain CMakeLists.txt"

MattSturgeon, 2016-06-13:
@inflex [this](/MattSturgeon/OpenBoardView/tree/mingw-linux-fix) is probably a good place to start.

clone that branch and remember to `git submodule update --init --recursive` then just install build dependencies and run `./openboardview.sh`

The reason is the slight-refac branch doesn't have @piernov's unix port, @pafmaf merged it into a different branch.

inflex, 2016-06-13:
@MattSturgeon thanks for that, running nicely now in linux.  Couldn't get json11 to build out of the box but I cloned it off the dropbox git repos and it all seems fine now with the appending of the LD_LIBRARY_PATH to make it load up the libimgui.

Thanks, now I can get back to fixing broken electronics :)

inflex, 2016-06-13:
@MattSturgeon  is there anything you need to have worked on for the linux release?  I do a lot of linux development, though I admit, it's all CLI tools/filters, rarely do I work within the GUI realm.

piernov, 2016-06-14:
Since I couldn't figure out how to properly make a DMG with CMake, I ended up creating an XCode project. Now you have a (sort of) redistribuable OS X DMG here: https://github.com/piernov/OpenBoardView/releases/tag/v0.3-osx1

MattSturgeon, 2016-06-14:
It would be nice to get to the point where we can use cake to generate any build system we need (msvs, makefiles, xcode...) automatically - after all that's what it's designed for. That way we only maintain one set of build system files.

Using CPack for DMG, RPM, DEB and exe installers would be a great plus too.

I've been struggling to work on this since I've not really worked with build systems properly before and I don't have access to a Mac or Windows machine currently

inflex, 2016-06-17:
@MattSturgeon  not sure if I'm doing this completely wrong (oh wait, that would be exactly what I'm doing) but I noticed that in the linux build it doesn't support the letter-based keyboard shortcuts.  For some reason the shift-key requirement for things like R N L C K shortcuts just doesn't work, it's almost like SDL is remapping them back to lower-case letters?   

[BoardView.cpp.zip](https://github.com/chloridite/OpenBoardView/files/320154/BoardView.cpp.zip)
I created a lower-case check with the uppercase in BoardView.cpp as well and now it works

MattSturgeon, 2016-06-17:
@inflex Yeah I've been noticing problems the shortcuts system (though I'm not sure its issues are linux specific).

Interestingly most of the shortcuts still don't work on my system, I suspect this may be because I have a UK keyboard layout.

By the way, I stuck your changes [here](/MattSturgeon/OpenBoardView/commit/745293c3511161308bfcf356cea91cb80c07298d) for safe keeping. If your interested I've also been doing some work and [playing around with the build system](/MattSturgeon/OpenBoardView/commits/cmake-changes) and started work on [embedding the assets](/MattSturgeon/OpenBoardView/commits/embed2).

inflex, 2016-06-17:
@MattSturgeon can't imagine it'd be a UK layout issue, at least I wouldn't have thought it would be as they're not generating different keycodes *scratches head*

Right now I'm being horribly kludgy and working on a recently-used history list, because I'm lazy and don't like trying to filter through my folders reloading boards I used 3 sessions ago :) 

Good luck with the build system, I think once that's pinned down and running smooth it should help a lot.  I still can't seem to get json11 to build as part of the script, have to do it manually, it's odd.

[image]

inflex, 2016-06-18:
@MattSturgeon  I've finished the file-history change and I've also made the window title show the currently loaded board file.  Small things I know but I find it helps me when I'm switching around a lot.

The bad thing right now is that, well, I'm not sure how to properly offer these changes in to git, guess I better get learning on this too (sorry, ancient old school from the days of UNIX RCS )

inflex, 2016-06-18:
@MattSturgeon have you noticed that OBV will chew up a lot of CPU when it's minimised/shaded?

I had to add an explicit sleep/do-nothing in main_opengl.cpp 
uint32_t w;

while(SDL_getevent(&event)){
....
}
w = SDL_GetWindowFlags(window);
if ((w & SDL_WINDOW_INPUT_FOCUS)==0) { usleep(50000); continue; }

After that I went from my system dragging down with OBV hitting 60~70% CPU back to insignificant.

inflex, 2016-06-24:
Sorry to keep bumping this thread, please let me know if I should make notifications elsewhere.

Anyhow, I've had a bit of a run of work here and have added numerous things I think might be useful for other users of OBV, including;

  * variably sized parts/pins display based on part geometry
  * Arbitrary angled bounding boxes around parts based on the minimal-bounding-box algorithm
  * Resistors/caps/diodes/inductors that aren't at 90/0 degrees are now displayed with more realistic angled box.

I still have to clean up the code, as it's been a real hack & slashfest while I sorted this out.

Attached is a screen shot demonstrating this.
[image]

kikef333, 2016-06-24:
@inflex would you make an executable file? thanks!! there´s a post were they added features of zoom in and out and dragging and dropping files, check them out and maybe put them together with your man! you rule!

inflex, 2016-06-24:
@kikef333 I'll have to see about the zoom in/out, since that's already there (using mouse and num-pad, oh that's right yes, I added pan/zoom with numpad a few days ago).

The drag-drop I'm not sure about yet as I think a lot of that will depend on the actual desktop-management system OBV is run in; however, I have already added the ability to load via the command line (along with the history of course).

While I can make a binary on my machine I'm not sure if it'll run elsewhere properly (linux).   Are you on Linux or windows?

inflex, 2016-06-24:
@kikef333 I just tested/checked, adding drag-drop support is a 4~5 line addition so that's a definite easy add on the SDL/Linux side.  Probably would want a confirmation dialogue though if you've already got a file open.

Anyhow, working out some issues with my minimum-bounding-box algorithm and I'll get back to adding the drag-drop.

kikef333, 2016-06-25:
@inflex https://github.com/charliebruce/OpenBoardView/releases/tag/v0.4

inflex, 2016-06-25:
@kikef333 finished it all up about 10 hours ago :)   About the only thing I need to change on my fork to match will be the filtering of *brd files in the file picker dialogue, other than that, it's good to go.

inflex, 2016-06-25:
@MattSturgeon how far apart is the linux fork from the Windows?  Is it practical/possible to keep them fairly parallel or do we think that it's going to degenerate in to two moderately separate builds?  

Seems like things had to be reworked a fair bit to get the build to happen, sharing mostly just the loading/board-interpreter core?

inflex, 2016-06-25:
Have registered #openboardview on Freenode IRC is anyone is wanting to have casual discussions without my posts clogging up the system

pafmaf, 2016-06-25:
I personally don't have a problem with IRC but maybe some of the potential devs aren't too familiar with it / wouldn't bother to set things up. 

So what about some alternative service like Gitter?

To sum up:
* more beginner friendly UI in Web browser
 - no external client needed
 - unread messages etc.
* login using your GitHub account
* history is easily accessible
* didn't try but it seems there's https://irc.gitter.im/ for interop
* free up to 25 users (but I doubt there will be more active users than that)

inflex, 2016-06-25:
I guess in this instance it was a case of greybeard habits, since I already have IRC on/open for other projects on Freenode.

MattSturgeon, 2016-06-25:
@pafmaf Gutter sounds good, the irc integration sounds nice too.

@inflex as it stands in the Unix port, things are split up according to @pafmaf's restructure. @piernov split things further (into unix.h, osx.h, opengl-stuff and dx9-stuff) and added CMake to get things working cross platform.

I have been thinking about re doing the port in a less fragmented way, but I don't have as much time (or code familiarity) as I'd like.

It would be nice to get @pafmaf's PR #14 into @chloridite's main branch as (if I'm not mistaken) this is the bulk of the code refractor the current port is based on.(?)

inflex, 2016-06-30:
@MattSturgeon any word back from @chloridite ?

MattSturgeon, 2016-06-30:
@inflex I still need to send the email.

I've been distracted, focusing on different ways to do the port and trying to get CMake to build a working VisualStudio project. I've got a [draft that uses GLFW instead of SDL](https://github.com/MattSturgeon/OpenBoardView/commits/port3), for example (though it looks like some of your changes are SDL specific?)

inflex, 2016-06-30:
Well, if the port is buildable across platforms then it shouldn't be a lot of trouble to rewrite stuff, since the *hard* part was mostly the math/algorithms; the final rendering really only called on fundamental poly/circle/rect generation calls.   There was the drag-drop stuff which I'm not sure if OpenGL handles.

It'd only take me a day or so to re-port all my mods, though yes, it'd have been nice if we worked out how to magically make them in the current form :)

piernov, 2016-10-30:
R7.3 has support for Linux (64bit RPM, DEB and compiling from sources), OS X (DMG for 10.9 and later I think) and Windows (32bit for Windows 7 and later).

closed 2016-10-30 by piernov