OpenBoardView/OpenBoardView #10: Asus boardview file format (.fz)

piernov, 2016-05-30:
Could you try to decode this format? (and if possible, add support for it)
Attached is a few boardview files (inside the fab/ folder) and the tool used to read them.
(If you get "License Error", just start it again. Use Ctrl+F10 to open file.)
[PCBRepairTool.zip](https://github.com/chloridite/OpenBoardView/files/289925/PCBRepairTool.zip)

pafmaf, 2016-06-08:
I believe .fz is some form of RC6 & gzip encoded text.

pafmaf, 2016-06-26:
```
#!/usr/bin/env python3

"""
This is for splitting up the resulting stream by decoding the RC6 file
with .fz ending.

The stream is split in two parts (potentially something like the contents 
followed by some content description) at an offset specified by the last 4 bytes
interpreted as an int32.
"""

from zlib import decompress as deco

buf = str()
with open("de-rc6-ed.bin", "rb") as f:
  buf = f.read()

# length stored as last 4 bytes LE (int)
descr_len = int.from_bytes(buf[-4:], byteorder='little')
(content, descr) = (deco(buf[4:-descr_len+4]), deco(buf[-descr_len+4:]))

print(str(descr, 'utf-8'), str(content, 'utf-8'))
```

pafmaf, 2016-06-27:
And here's the complete thing without the key, since I don't know it it's okay to share it here.

https://gist.github.com/pafmaf/cf7f44de76b65a5edb130fa928697bea

inflex, 2016-11-02:
FZ, including non-encoded and those using the Units:millimeters now supported as of R7.3 merge.

Is this closable now?

piernov, 2016-11-02:
Same as ".ASC", we don't use all the information available in the file but it renders correctly for a basic usage.

MattSturgeon, 2016-11-02:
Any additional .fz stuff can always be handles in new bugs

closed 2016-11-02 by MattSturgeon

RavetcoFX, 2017-05-26:
@pafmaf How did you go about grabbing a key?

pafmaf, 2017-05-27:
If you're asking how to get it, check on IRC.

my6010, 2019-07-20:
Hello, have you decoded the FZ file structure?

pafmaf, 2019-07-20:
> Hello, have you decoded the FZ file structure?

@my6010 You mean whats described in? https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/FileFormats/FZFile.cpp 

Not sure I get your question.

my6010, 2019-07-22:
Do you have contact information for dating software? For example, MSN! Some questions would like your help!

pafmaf, 2019-07-23:
Not sure anybody still uses MSN.
Why dont you simply describe what you are trying to achieve?

piernov, 2019-07-23:
Pick up someone on Github…? Anyway, there's also an IRC channel #openboardview@irc.freenode.net , but we cannot really provide individual support. Any bugs and feature requests as well as (highly welcome) contributions should be submitted to this issue tracker.

my6010, 2019-07-24:
[UN66_1..txt](https://github.com/OpenBoardView/OpenBoardView/files/3424751/UN66_1.txt)
I want to know how to convert TXT source code into FZ format and turn it on for normal use. At present, I already know that someone can complete this conversion, but do not provide methods!

pvtinflex, 2019-07-24:
Would be easier to write an importer for that format than try convert to FZ; looks nicely structured.  What program generated it?

my6010, 2019-07-25:
> Would be easier to write an importer for that format than try convert to FZ; looks nicely structured. What program generated it?

The TXT source code is captured on the server's online viewing by using the package capture plug-in. It is not clear what program or method is used to convert it to FZ for the time being. This is the purpose of my asking for help!

my6010, 2019-07-28:
> > Hello, have you decoded the FZ file structure?
> 
> @my6010 You mean whats described in? https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/FileFormats/FZFile.cpp
> 
> Not sure I get your question.

I mean, I need a copy of the internal data structure of the decoded FZ file. Do you have it?

piernov, 2021-06-30:
@FrannDzs what is your request?

Awaketo, 2022-02-16:
Sorry for this newbie question, what is IRC?

> If you're asking how to get it, check on IRC.

piernov, 2022-02-17:
The IRC channel has been abandoned, someone indeed posted the FZ key badcaps.
https://www.badcaps.net/forum/showpost.php?p=1085662&postcount=103

JL01152, 2023-01-29:
Are there any updates as far as getting .fz files to be able to open? I have an Asus Z97-AR  .fz file I'm trying to open and I get the following error when trying:
[image]