OpenBoardView/OpenBoardView #11: Custom file format / wrapper.

pafmaf, 2016-05-31:
This is a suggestion / RFC thing.

Most files I work with come as two distinct, weirdly named files, i.e. `<weird_name>.brd` layout and `<weird_name>.pdf` schematic.

While working on the code I noticed I have no good way of storing annotations (e.g. component aliases) and when you want to find files quickly, it'd be cool to have more context/details on those files, like what product they possibly apply to.

Another thing is having background images for both sides, that's two more files for just one thing (unless you merge them etc.), so instead of having those simply in one folder I plan on implementing a 'custom file format' based on a zipped folder, something like: 

```
<weird_name>.zip>
| - ! board_info.json  # friendly name, model name, zip structure, other stuff
| - thumbnail.jpg  # thumbnail for some explorer thing?
| - annotations.json  # containing alias component names, highlights, text annotations
| - favourites.json  # possibly a list of favourite nets, components?
| - ! <weird_name>.(*.brd, *.bla)  # original board view file
| - <weird_name>.pdf  # schematics PDF for possible integration later on (MuPDF?)
| - hires_img/  
  | - top.jpg
  | - bottom.jpg
``` 

(Elements with a `!` have to be present, the rest is optional.)

So it's a singe file containing all info, that may be hand-edited, customized and shared with collegues.

Any ideas on this?

drewrox2009, 2016-06-05:
I absolutely love this idea! Id love to help any way i can. I am not proficient in any programming languages, but id be happy to brush up. I can definitely help with compiling aliases and grabbing info. I think there is definitely room for this software to grow. Correct me if i am wrong but this software is written in c++. Ill Go through some of my old books.

inflex, 2016-06-21:
@pafmaf it's a good idea (I think ;) ).  I think just staying with zip is one of the better options, easy to get existing libs that handle it all and then we make OBV simply react/manage based on the internals. 

The PDF integration is definitely something I'll be chasing up as I was already thinking of installing a small listening socket on OBV and then modifying something like evince to send component/net requests to OBV when you select them in the PDF, but an integrated PDF would be far saner :)