OpenBoardView/OpenBoardView #12: So, not so much an issue..

doglover129, 2016-06-03:
Not so much of an issue, but rather a suggestion. Is there any way or any chance this will become multi-platform? I am primarily a Mac user, and would love to see this program there too! (Don't worry, I already downloaded it for my Windows partition). Maybe a java based program instead, or at least a Mac or Linux version in the future?

I know this program is new, but it would be cool!

piernov, 2016-06-03:
Duplicate, see #9 . I already ported it to Linux and OS X and it seems to work without problem, given you know how to build it "by hand" (well, my build.sh and the CMake scripts do it for you but you still have to install the needed dependencies, with homebrew on OS X for example).

Also note that Java is not *the* solution for multi-platform apps. This app was not really difficult to port on Linux&OS X. We're only running into packaging issues because we don't know how it works on OS X with CMake.

rfranco2001, 2016-06-06:
It works perfectly fine under Wine for OS X (and I guess Linux will be fine as well).

pafmaf, 2016-07-05:
Duplicate, as piernov said.

closed 2016-07-05 by pafmaf