OpenBoardView/OpenBoardView #13: Error opening file

amalic, 2016-06-03:
I get an error message when trying your software. After looking for brd files i found one on github, which resulted in the error described.

https://github.com/sparkfun/Si4703_FM_Tuner_Evaluation_Board/blob/master/Hardware/SparkFun_Si4703_Eval.brd

stray77, 2016-06-03:
This is an eagle cad file that just happens to share the same extension as the files we use.  They are not intended to be compatible.  I'm guessing you should close this issue.  Here is an example of a file that does work. 
[LB#820-2530 Macbook_13_A1278_K24.zip](https://github.com/chloridite/OpenBoardView/files/298189/LB.820-2530.Macbook_13_A1278_K24.zip)

amalic, 2016-06-09:
thanks for the infos

closed 2016-06-09 by amalic

MattSturgeon, 2016-07-05:
If anyone is actually interested in eagle cad files, please open a new issue