OpenBoardView/OpenBoardView #15: Expanding Use Case

merlinacious, 2016-06-09:
I found this via the Louis Rossmann youtube video and was wondering if this could be expanded to a broader use case of working with electrical diagrams from say Motorcycles, Cars etc.

I don't know if those manufacturers publish .brd files but that would be an awesome application for this program.

inflex, 2016-06-13:
I'd imagine it'd be more something up to the people who get a hold of the schem/layouts as opposed to this software?  btw, also here because of Louis :+1:

MattSturgeon, 2016-07-05:
Since we don't know if or what files are published by the other industries, I'm adding "more info needed"

piernov, 2016-10-30:
Closing because it's an old issue and OpenBoardView is limited to working with boardview files. If those manufacturers provide boardview files in a different format, we would be glad to implement it if you open a new issue.

closed 2016-10-30 by piernov