OpenBoardView/OpenBoardView #20: Hi! amazing! great job! could anybody make the executable with the zoom feature?

kikef333, 2016-06-20:
and is it possible to make it on dmg so we can run it on mac? again thanks!!!! you rule!!!

slayer200230, 2016-06-20:
i was just gonna ask how to do it ourselfs, i managed to update the files. and combine both the file name updates and the zoom/move updates, but have no idea how to compile it into an .exe

charliebruce, 2016-06-20:
@slayer200230 Just realised that I hadn't responded to your question re compiling. To compile an .exe for yourself, you can download a copy of Visual Studio 2015, clone a copy of the repository using Git, and open the project using that. Then you would select Build->Build Solution, and it should build the executable in a "Debug" or "Release" folder inside your local repository. It's a bit out of scope to walk you through it in much detail, but there should be some guides to Visual Studio online if you're struggling with anything.

slayer200230, 2016-06-21:
@charliebruce love it, ty ty, the updates are wonderful!

kikef333, 2016-06-21:
@charliebruce the zoom in and out works with what buttons? thanks again!

kikef333, 2016-06-21:
@slayer200230 if you manage to make zoom in and out and compile please let me know! :)

slayer200230, 2016-06-21:
@kikef333 the + and - keys up by your number keys, zoom in and out, move with arrow keys or w,a,s,d
ofc make sure you are using @charliebruce 0.4 release that was posted above, the 0.4 release also has the drag and drop feature as well as the directory/file name

kikef333, 2016-06-24:
working flawless!

MattSturgeon, 2016-08-17:
Should be fixed in 4d245d9

closed 2016-08-17 by MattSturgeon