OpenBoardView/OpenBoardView #22: exe wont launch on two systems

lerocm, 2016-06-22:
OpenBoardView.exe wont launch on the two systems below. After double clicking the exe, the hourglass icon appears for a brief moment then nothing happens.

Ist system: MSI AP1920 Windtop running Windows 7 Home Premium 64 bit OS, CPU Atom D525 , Integrated graphics Intel GMA 3150, Avira Antivirus running, Zone Alarm, DirectX 11

2nd system: MSI Wind U100 Netbook, WinXP Home Edition SP3, CPU Atom N270, Integrated Graphics Intel GMA 950, Avira Antivirus, Zone Alarm, DirectX 9.0c

All other board viewers run, including the TopTest/Landrex versions and NeoBoardViewer

slayer200230, 2016-06-22:
1 what ver you trying to run? 2 works on win 10 and linux just fine, avira isnt a problem as i have it.

piernov, 2016-06-23:
Just tried on an Asus 1201NL (Atom N270) with Windows 10 and it works, so it's not an issue with the CPU. But on this machine the GPU is an NVidia ION. It might be a software issue: broken DirectX 9 or graphics driver for example. You should try to update/reinstall both.

LarryOffley, 2016-06-23:
I also have a Windows 7 64bit machine that it won't open on. Intel Core 2 Duo With Intel Q35 Video Card. Dual Monitors. It runs on every other machine in the shop but mine. I have reinstalled Direct X done a SFC scan nothing else seems to have a problem. I wonder if some other software is causing a conflict.

slayer200230, 2016-06-23:
@LarryOffley do you happen to have zone alarm? i think it has to be software. what kinds of antivirus do you have?

LarryOffley, 2016-06-23:
Avg but that is on machine that it does work on.

I have a lot of open source stuff might be a library conflict I don't know
yet.

Larry

charliebruce, 2016-06-23:
Try this build out:

[OpenBoardView-ErrorMessages.zip](https://github.com/chloridite/OpenBoardView/files/330650/OpenBoardView-ErrorMessages.zip)

Hopefully it'll show an error message rather than just silently exiting.

LarryOffley, 2016-06-23:
Having the error is very useful

Error:Failed to create direct 3D device.

Interesting I haven't had any other programs fail. I will try reinstalling
my video drivers anyways.

Larry Offley

lerocm, 2016-06-23:
Thanks charliebruce. On both systems the error message is "Failed to create Direct3D device".
DxDiag shows no problems via "test directdraw" and "test direct3D" on the N270 winxp system. DxDiag doesn't have a test option in win7.
I agree its probably a software issue. Because its occurring on two such different systems (xp and win7), it must be something they have in common. Like a missing .net framework. However both systems do have closely related crappy Intel integrated graphics GMA 3150 and 950. Openboardview might not play nice with these graphics. They have limited features and limited openGL implementation.

LarryOffley, 2016-06-23:
Mine is a Intel Q35 Chipset Video Card wonder if these older Intel cards are just missing a feature the program is using. Still checking if there is a newer driver. otherwise I'll throw a different card in the box

charliebruce, 2016-06-23:
I suspect the graphics chip doesn't support all of the features the program wants. Try this build now and let me know what happens:

[OpenBoardView-SoftwareFallback.zip](https://github.com/chloridite/OpenBoardView/files/330794/OpenBoardView-SoftwareFallback.zip)

LarryOffley, 2016-06-23:
Fail to create accelerated blah blah falling back to hardware. Click ok and
it opens to Black box with Open Board Viewer at top.

If you want for testing I'll leave my machine like it is for a bit. You
caught me just before I was going to replace the video card with and old
nvidia 7600 just to get it to work on my work computer. (I don't usually
need much video on this machine)

charliebruce, 2016-06-23:
Does the program work correctly after that warning though?  Ie you can open a board file, click things, zoom in and out?

LarryOffley, 2016-06-23:
no menu. just black with normal program boarder. no keys seem to do anything

lerocm, 2016-06-24:
I have identical experience to that described by LarryOffley.

LarryOffley, 2016-06-24:
I put a Nvidia 8400 card in and now it works perfect. Must just be the
older intel cards are missing support for a feature.

Love the program by the way. So much better than the other ones.

lerocm, 2016-06-28:
Unfortunately the two systems I have don't allow for an external graphics card, only integrated graphics available. I should add, using the OpenBoardView_SoftwareFallback build, although you only get a black box with no menus or graphics, ctrl-o will bring up a file selection box and the selected file name is displayed in the top boarder of the box.
[image]

MattSturgeon, 2016-07-04:
Out of interest, does [this opengl port](https://github.com/chloridite/OpenBoardView/files/346245/OpenBoardView-0.3.0-win32.zip) work?

(You'll need to unzip the installer and then run the setup program)

up-tick, 2016-07-05:
The program starts, but will not open eagle .brd files. I installed the DX files from MS.
64bit Win7homepremium, simple video card. Radeon 6450.
The 32bit program complains about a missing MSVCP140D.dll.

MattSturgeon, 2016-07-05:
> The 32bit program complains about a missing MSVCP140D.dll.

Sounds like it's missing a debugging lib.

Try this non-debugging build: [OpenBoardView-glfw-port-win32.msi.zip](https://github.com/OpenBoardView/OpenBoardView/files/347896/OpenBoardView-glfw-port-win32.msi.zip)

The opengl port is far from finished and being changed rather frequently, but I'm interested if it will play nicer (or maybe worse) with these graphics drivers.

lerocm, 2016-07-06:
Yes thank you, the non-debugging build: OpenBoardView-glfw-port-win32.msi.zip with inbuilt installer works fine for me. 

One tip - In win7 and possibly win 8 and 10 you need to enable aero for opengl to work properly. I like using the Windows Classic theme with all bells and whistles turned off to speed things up, but this stuffs with opengl. Something to do with Vsync. Found this out years ago trying to run half life/counterstrike and other steam platform stuff.

I've attached a .brd file that should work with openboardview but throws up an error message when trying to open it. It opens with neoboardview and Top Test Boardview R4. I'd be grateful if someone could run an eye over the file and see if it can be easily modified to open. Otherwise no need to waste time on it. Thanks in advance.
[Dell_Inspiron5150_LA1682_brd.zip](https://github.com/OpenBoardView/OpenBoardView/files/348929/Dell_Inspiron5150_LA1682_brd.zip)

MattSturgeon, 2016-07-06:
> One tip - In win7 and possibly win 8 and 10 you need to enable aero for opengl to work properly.

Interesting, may add that to the README when the port lands - thanks!

> a .brd file that should work with openboardview but throws up an error message when trying to open it. It opens with neoboardview and Top Test Boardview R4.

Would you mind starting a new issue for it? Sounds like this could affect some people.

up-tick, 2016-07-06:
I did not try Aero interface, but I'm getting a MSVCP140.dll missing error,  which does exist in some 12 places.

up-tick, 2016-07-06:
I did try Aero, no difference.

charliebruce, 2016-07-06:
Try installing Microsoft Visual C++ Redist :
www.microsoft.com/en-us/download/details.aspx?id=48145

up-tick, 2016-07-06:
I did, same deal. I run win7 in "test mode" "Build 7601"

piernov, 2016-10-30:
Closing because Windows binary is built with MinGW as of R7.3 and we don't support Windows XP anymore.

closed 2016-10-30 by piernov