OpenBoardView/OpenBoardView #26: New logo

MattSturgeon, 2016-07-10:
With new file formats getting closer to being ready, a more generic logo should be designed. Additionally it would be good to have something that scales well to different resolutions.

The logo should be designed with SVG to allow exporting to different sizes and formats easily, as well as making any future changes easier.

See #27 for working on filetype icons.

MattSturgeon, 2016-07-13:
I've come up with two drafts in inkscape, based on the current icon and using the same colour scheme.

Version one has rather spaced out text
[image]

While version two has larger, more squashed together text
[image]

I did try using a circular background, however with the text and the border, it just didn't look right.

I'm up for suggestions on more abstract themes though - perhaps some kind of symbolism for a board or some kind of mascot?

devgiant, 2016-07-14:
Can I assume that this is the color palette we are aiming for, or are we thinking in changing the colors?

http://www.colourlovers.com/palette/4290450/Open_Board_Viewer

I'll see if I can help with the artwork part.

MattSturgeon, 2016-07-14:
Regarding colour palette, so long as it "feels" like OpenBoardView, I don't think people will mind. Might be worth putting together a small collage from screen shots of OBV.

Any help or info is welcome. Also need to think of something for the various file types in the works.

If possible please keep master copies as SVG to enable easy modification and scaling.

devgiant, 2016-07-16:
For the icons I think we can use something like this:

[image]

[OBV_BRD_icon.zip](https://github.com/OpenBoardView/OpenBoardView/files/367427/OBV_BRD_icon.zip)

And if you guys think it's ok, I can create one for each format and export them in all the sizes and formats needed.

For the main application icon/mascot I'm still trying to create something that is acceptable, all the previous attempts were unappealing.

MattSturgeon, 2016-07-17:
> For the main application icon/mascot I'm still trying to create something that is acceptable, all the previous attempts were unappealing.

Yeah, coming up with something nice is hard. Especially as we don't really know what we want!

---------------------------

I think those file icons look rather good. Thanks for taking the time to make these.

Personally I like the rounded versions more than the square edges. With the text, I'm torn between the versions with and without the background, though I feel like I'm leaning towards having the background.

I've played around a little, and rather like the look of the [Courier Prime] typeface:

[image]

[OBV_BRD_icons_2.zip](https://github.com/OpenBoardView/OpenBoardView/files/367602/OBV_BRD_icons_2.zip)

I think using the Courier Prime in bold, with a custom circle rather than the standard `.` character seen above, could look quite good. I also corrected the point alignment for the folded corner as it was about half a pixel out.

[Courier Prime]: http://quoteunquoteapps.com/courierprime/

inflex, 2016-07-17:
I like the fundamental look/feel,  would like to add on top something like this (done in Inkscape)

[image]

devgiant, 2016-07-17:
The font I used was Monserrat, you can find it here:

https://www.fontsquirrel.com/fonts/montserrat

I chose that font because of the license, so that we could use it without any kind of problem, but also because I think it fits ok.

Courier prime has the same license, so we can use it instead, if you guys prefer.

https://www.fontsquirrel.com/fonts/courier-prime

I've tried to add something similar to what @inflex sugested:

[image]

[OBV_BRD_icon_.zip](https://github.com/OpenBoardView/OpenBoardView/files/368012/OBV_BRD_icon_.zip)

Tell me what you think and if I should change to Courier Prime.

pafmaf, 2016-07-17:
Good job. I like the top-right one best, I think that this font seems a *tad* more readable when the icon is scaled down. The 'decal' is something that makes it more unique, recognizable + it gives a hint as to what the file is for.

What I think would be an improvement is a light border around the whole thing, since some people on Linux for example are known to have very dark GTK themes. So just some 1px white contour would make it stick out from the dark file manager background while still invisible on preset Finder/Explorer background.

MattSturgeon, 2016-07-17:
> I think that [the top-right one's] font seems a tad more readable when the icon is scaled down

I've been meaning to take a look at the icons at different sizes. I'm not totally shocked that the sans-serif font may look cleaner at small sizes. I know @inflex has said he prefer's @devgiant's choice of font over mine :grinning: 

> So just some 1px white contour would make it stick out from the dark file manager background

I'm not sure about a 1px boarder, but maybe a "glow" style white gradient could serve the same function while being a bit subtler?

-----------------------

I haven't checked these SVGs' yet, but if possible I would prefer that the file extension be an SVG text field rather than a path; this would simplify expanding to other filetypes in the future.

piernov, 2016-07-17:
The one made by @inflex is nice but the traces are too thick.

About @devgiant : For me it's the opposite, when zooming out I find the bottom left more readable, but when zooming in I prefer the top right.
Maybe you should try removing the dot and making the text a bit bigger?

pafmaf, 2016-07-17:
>  a "glow" style white gradient

But only if all file managers support that. (Or just for some hi-res version?)

> zooming out I find the bottom left more readable, but when zooming in I prefer the top right

Looks like we need different versions for different sizes then?

devgiant, 2016-07-17:
@piernov , to increase the text I would have to increase the background as well.

@MattSturgeon I added a white 1px outline but it doesn't fit too well. We can improve contrast by adding a lighter outline color that isn't too extreme, something like this:

[image]

[OBV_BRD_icon.zip](https://github.com/OpenBoardView/OpenBoardView/files/368125/OBV_BRD_icon.zip)

@pafmaf a white glow is a little outdated look, so I don't think we shoud consider it, but if you guys think is the way to go we can try it. Either way we will need different sizes and different logos for different uses.

inflex, 2016-07-17:
@devgiant I like @piernov 's suggestion about dropping the leading '.' but perhaps that breaks standard convention?

Only trouble I have with the black traces (and thin) is that they're hard to see/distinguish when zoomed back,  yes, my traces were very thick though I was only just trying to give the idea :+1:   Also, given the already dark colour of the folder, could we see what it's like with white traces too?

MattSturgeon, 2016-07-17:
> We can improve contrast by adding a lighter outline color that isn't too extreme, something like this

Agreed. That looks much better than I was expecting, no "outdated" glow needed :smile: 

------------------

Regarding scaling, we need to play around with the text size and the thickness/amount of the background traces so that the icon remains clear and attractive at all reasonable sizes.

If neccessary we could have a "small" "medium" and "large" version of the SVG where details can be tweaked or replaced as needed; though it'd obviously be easier to have one icon that just works for all sizes.

inflex, 2016-07-17:
This is sort of what I was thinking of perhaps?

[image]

devgiant, 2016-07-17:
I had a few iterations done before I saw your comment. Tomorrow I'll try to redo them accordingly.

[image]

[image]

[OBV_BRD_icon.zip](https://github.com/OpenBoardView/OpenBoardView/files/368180/OBV_BRD_icon.zip)

devgiant, 2016-07-26:
Sorry for the delay but time has been an issue to me lately.
@inflex, After some attempts I always turn back to your version as it is way better than my new ones, so I think we should keep it and move on to the logo/mascot. 
If anyone has a suggestion about it, I'm all ears. :)

inflex, 2016-07-26:
@devgiant well it's been a combined effort all around.  I'll upload the SVG just shortly.

inflex, 2016-07-26:
FIle attached.  Note that there's a tiny protrusion at the bottom due to myself not quite knowing the best way to cap the trace line at a 45' angle.

[OBV_BRD_FILE_icon.zip](https://github.com/OpenBoardView/OpenBoardView/files/385040/OBV_BRD_FILE_icon.zip)

MattSturgeon, 2016-07-27:
@inflex I've edited your comment with a fixed SVG. I aligned the top right fold, and cut the bottom trace to align with the bottom of the icon.

I also replaced the path-text with actual text to aid future editing. (I included the font otf in the zip); exported icons wouldn't need this.

MattSturgeon, 2016-07-27:
Heres my slightly messed up attempt at adding the border back in: [montserrat.zip](https://github.com/OpenBoardView/OpenBoardView/files/385436/montserrat.zip)

I copied the icon, enlarged each part by 2px and repositioned at -1px then used _union_ to create a path. For some reason the curves look a little messed up to me...

devgiant, 2016-07-27:
@MattSturgeon All your files open like this:

[image]

I don't know if its an issue with the svg file or if it's just me, can somebody test it? Thanks.

MattSturgeon, 2016-07-27:
@devgiant Working fine for me in Inkscape; here's a plain SVG version: 
[OBV_BRD_FILE_icon_with_border.plain.zip](https://github.com/OpenBoardView/OpenBoardView/files/385462/OBV_BRD_FILE_icon_with_border.plain.zip)