OpenBoardView/OpenBoardView #28: Basic controls config file

MattSturgeon, 2016-07-11:
It would be good to be able to add configurability to the controls, as this seems to be the main thing people disagree on.

Things I think should be user configurable:

- Keys used for different things
- Speed of various movements

For the speed adjustments, I am thinking we should add global multipliers like `uerZoomSpeed` and `userPanSpeed` that default to `1.0f`, we then simply multiply against our internal multipliers.

For the keys, I am thinking of an interface along these lines

```
// After parsing config file, overwrite default with "<CTRL>O"
actions[OPEN_FILE].setKey(KEY_O).addModifer(KEY_CTRL); 

// Check if action is being pressed
if(actions[OPEN_FILE].pressed()) { /* Open a file */ }

// Create a menu item
ImGui::MenuItem("Open", actions[OPEN_FILE].getKeyComboStrPretty())
```

For the config file, I'm thinking basic `key=value\n` pairs. Each keycombo value could follow a format such as `<modifiers>ACTION_KEY`. We could convert the keystring into an index using a translation/hash table.

PierreKircher, 2017-07-12:
+1 for that there are a few ( like me ) who dont use qwertz/qwerty  .. custom bindings would be nice to have

piernov, 2021-06-21:
I believe the keyboard controls configuration is available as of the latest development branch, and the custom zoom speed has been available for a long time.

closed 2021-06-21 by piernov