OpenBoardView/OpenBoardView #31: 'Freeze' after opening file (Win file dialog preview bug)

pafmaf, 2016-07-12:
This is a weird one. When openening a file, if the file dialog has the _preview panel_ open and it occasionally has to preview files of some sort (not all file types will trigger this, but apparently `.ini` do this), OBV will not refresh the window until it is resized or minimized & restored.

This, most likely, isn't a bug in OBV, but more a Windows issue, - just in case someone's having similar problems: resize or minimize the window after opening a file with the Windows file dialog.

proof: [Link To GIF](https://cdn.pbrd.co/images/aA0ZiH43A.gif)

MattSturgeon, 2016-07-12:
I [tried using the new IFileDialog API](), but other than having a different interface (might even be nicer, at least for filetype filtering) it seems the same dialog is used as windows translates the old interface to use the new dialog if there's no incompatible usage.

An alternative might be forcing windows to use the old XP-style dialog, but hopefully there's a better solution...

pafmaf, 2016-07-12:
> An alternative might be forcing windows to use the old XP-style dialog, but hopefully there's a better solution...

Well, this is practically a `wontfix` to me anyways. Simplest workaround is to not have the preview open if it really is a problem. If the freeze does occur, you can still minimize.

It's just good to know that this is most likely not an issue with this software but with the crap called Windows, before s.o. starts poking at the GL or DirectX part.

That being said, what might be interesting is whether or not that freeze may have other causes, e.g. on changing monitor resolution or moving between virtual desktops- then it's time to investigate.

mrxordi, 2016-08-30:
They should use another file manager(QT, Wx), however I'll dive deep into it and maybe find solution. I know there is hooking possible into standard dialog to replace or turn off totally preview.
We can do "fake resize" the window after every file open to force the window to refresh for now.

Can you tell me exactly how to reproduce the freeze?

piernov, 2016-08-30:
We didn't want to add an unnecessary dependency (especially on the Windows build where it makes things more complicated) and we try to use the OS' API when available. On Linux though, the open file dialog was implemented using GTK+.
What could be interesting is to see if the issue is still present in the inflex' Win32 OpenGL build: https://github.com/inflex/OpenBoardView/releases/tag/R4

mrxordi, 2016-08-30:
There is another issue for me where after file opening the board is moving with the mouse pointer.
There is something wrong with message processing because it doesn't release mouse button after file open.

inflex, 2016-08-30:
@mrxordi, just add this after the  filename = show_file_picker(); line

```
			ImGuiIO &io           = ImGui::GetIO();
			io.MouseDown[0]       = false;
			io.MouseClicked[0]    = false;
			io.MouseClickedPos[0] = ImVec2(0, 0);

```

If you're working on the master branch then you'll probably end up duplicating a lot of the work to fix small bugs that's already been done in the inflex-ui-features fork.

MattSturgeon, 2016-08-31:
@mrxordi I have pushed @inflex's above fix in 7926ef89626a2f719fdefafbd123024dfb6e6c2e

Normally it'd be good if you make a new isse if it is not the same problem, but thanks for taking the time to let us know. :smile:

piernov, 2016-10-30:
Is this still an issue on R7.3?

romanstingler, 2020-08-27:
@piernov as there is no response for such a long period I would close it.