OpenBoardView/OpenBoardView #33: Use file association and default programs

MattSturgeon, 2016-07-17:
Closely related to #27, we should set up OpenBoardView to be associated with `.brd` files and ask to be the default application.

On freedesktop.org systems this involves registering the `.brd` `application/x-brd-board` mimetype (as we will do for icon association anyway) and then list that mimetype in the `.desktop` file. Once we run `update-mime-database` and `update-desktop-database` The system will know to use `OpenBoardView.desktop` for `.brd` files.

On windows the relavent docs are [here](https://msdn.microsoft.com/en-us/library/windows/desktop/cc144156(v=vs.85).aspx). TODO: read them some more and edit this line.

IIRC on OSX the process for associating filetypes with applications is the same as setting the icon.

-----------------------------

Once the system knows to use OpenBoardView for `.brd` files, we still need to actually open the file the system gives us. This will likely just be a case of passing `argv[1]` (or `lpCmdLine[0]` if we are still using `WinMain`) to our open file method.

piernov, 2016-10-30:
It's done for Linux (with GTK) but not for other platforms in R7.3.

piernov, 2016-10-30:
(sorry, misread that, forget the previous notification)

PiezPiedPy, 2017-08-15:
> we still need to actually open the file the system gives us. This will likely just be a case of passing argv[1] (or lpCmdLine[0] if we are still using WinMain) to our open file method.

I'll have a look at adding this in a few weeks if you don't mind, its a feature that would make my life a tiny bit easier when opening files 😉

piernov, 2017-08-15:
Done already a while ago. See OpenBoardView/OpenBoardView@0cc08bd361abeceaf3c9a00bdd59e52ca13557c4.