OpenBoardView/OpenBoardView #36: New file format (?) .bv files

inflex, 2016-07-23:
Picked up this file today when trying to repair a board.  From the initial looks of it it's actually an MS Access / JET database ( even 'file' reports it as such ).

Anyone encountered this before?  Any prospects of getting a decoder written?

Hopefully there's a MDB/JET engine with suitable licence (MIT/BSD etc), and then actually pulling the data out could be interesting too.

[BV.zip](https://github.com/OpenBoardView/OpenBoardView/files/379337/BV.zip)

pafmaf, 2016-07-23:
> From the initial looks of it it's actually an MS Access / JET database ( even 'file' reports it as such ).

That's right.

```python
"""
Reads some data from BV board view file.

Run in PyWin.

Parts stolen from:
    http://stackoverflow.com/questions/25951912/oppposite-workings-of-oledb-odbc-between-python-and-ms-access-vba
"""

import win32com.client
import pyodbc

def ado(query):
    '''
    connect with com dispatch objs
    '''
    conn = win32com.client.Dispatch(r'ADODB.Connection')
    DSN = ('PROVIDER = Microsoft.Jet.OLEDB.4.0;DATA SOURCE = ' + db +  ';')
    conn.Open(DSN)

    rs = win32com.client.Dispatch(r'ADODB.Recordset')
    strsql = query
    rs.Open(strsql, conn, 1, 3)
    t = rs.GetRows()
    conn.Close()
    return t

queries = {
        'layout' : "select * from Layout;",
        'pin' : "select * from Pin;",
        'nail' : "select * from Nail;"
    }

if __name__ == '__main__':
    db = 'D:\\tmp\\MBP820-2054.bv'
    out = 'D:\\tmp\\'

    results = {}
    for name, q in queries.iteritems():
        results[name] = ado(q)

    for name, data in results.iteritems():
        with open(out+name+'.data', 'w') as f:
            for line in data:
                f.write(str(line) + '\n')
```

Contents are here: https://gist.github.com/pafmaf/f369a03a4a8c2e9e1dc3bde0ab055393

inflex, 2016-07-24:
Thanks for that @pafmaf .  Seems at least fairly straightforward, though the next step is finding a suitably licenced C++  interface. I suppose worse case scenario, call the above python script and consume its output.

inflex, 2016-07-24:
There's this so far that seems to be my biggest starting point.  It's GPL2 but it also has LGPL options, so that should be compatible.    If we have to roll-our-own, that shouldn't be too bad, won't be the first Microsoft format I've had to implement a decoder for (though I don't relish the idea of implementing even a partial SQL interpreter) :\

http://mdbtools.sourceforge.net/

inflex, 2016-07-24:
Under linux, I found if you install mdb-tools,  you can get a dump as well;

mdb-export sample.bv Layout > raw
mdb-export sample.bv Nail >> raw
mdb-export sample.bv Pin >> raw

pafmaf, 2016-07-24:
> It's GPL2 but it also has LGPL options

No problem here I guess:

> Files in libmdb, libmdbsql, and libmdbodbc are licensed under LGPL

inflex, 2016-07-24:
Next issue is, it'll be a different integration between windows and linux (and OSX?).   Am half wondering if it might be simpler to invoke a converter that uses the mdb-tools for a once-off conversion to an existing supported format?

inflex, 2016-07-24:
Working on a C/C++ implementation for linux at the moment using the libmdbsql (LGPL).

For windows, as you mentioned @piernov  , we just can use the native facilities there.

inflex, 2016-07-24:
Ended up writing a small tool which is a modification of the mdbtools 'mdb-export' source.

Attached is the modified source and the resultant output text dump.

Because mdbtools uses glib  it's a bit of a pain to try integrate it in to OBV for linux and I'm somewhat disinclined to taint OBV with glib.  At least on Windows it should be less drama.
[bv2raw.zip](https://github.com/OpenBoardView/OpenBoardView/files/380322/bv2raw.zip)

inflex, 2016-07-24:
Have added my preliminary .bvr decoder on my fork ( https://github.com/inflex/OpenBoardView/commit/bf400ed2dc6e430d129a5fbb5a388e85a9162571 ) 

Far from perfect yet, but it's a start.  You can get the raw dumped .bvr file at http://ctpc.biz/sample.bvr

[image]

inflex, 2016-07-24:
Seems there's an issue with the LOAD_STR() macro causing it to gobble up the top/bottom field when I'm using the TAB (\t) delimiter.

Soon as I changed it to a double-space the board comes out a lot cleaner ( see attached ).  Getting late though, 02H37, Monday morning customers ahead, best I get sleep :\
[image]

inflex, 2016-07-25:
Delimeter issue solved, was actually the LOAD_STR() macro that wasn't detecting the terminating \0 on the line.

corbolais, 2016-12-20:
Modified bv2raw source now available in repo, see [pr](https://github.com/OpenBoardView/OpenBoardView/pull/64).
The utilities/bvconv.sh script is yet to be cleaned up/handled properly. bvconv.sh was not creating valid BVRRAW_FORMAT_1 output.

PiezPiedPy, 2017-09-06:
Hi peeps
I've just used the utilities/bvconv.sh script to convert a .bv file but when the .bvr file is loaded into OBV all parts are only on the Top layer/side, any hints on what I need to edit in the .bvr file to fix the issue.

Here is an example of the lines for one of the parts in the Pins descriptions.

```
<<Pin>>
Part	TB	Pin	Name	X	Y	Layer	Net
"Q7960"	"(T)"	1	"1"	2.92300010e+00	2.72099996e+00	1	"1V05S0_SW"
"Q7960"	"(T)"	2	"2"	2.97300005e+00	2.72099996e+00	1	"1V05S0_SW"
"Q7960"	"(T)"	3	"3"	3.02300000e+00	2.72099996e+00	1	"1V05S0_SW"
"Q7960"	"(T)"	4	"4"	3.07299995e+00	2.72099996e+00	1	"1V05S0_TG"
"Q7960"	"(T)"	5	"5"	3.07299995e+00	2.91499996e+00	1	"PPBUSB_G3H"
"Q7960"	"(T)"	6	"6"	3.02300000e+00	2.91499996e+00	1	"PPBUSB_G3H"
"Q7960"	"(T)"	7	"7"	2.97300005e+00	2.91499996e+00	1	"PPBUSB_G3H"
"Q7960"	"(T)"	8	"8"	2.92300010e+00	2.91499996e+00	1	"PPBUSB_G3H"

```

PiezPiedPy, 2017-09-06:
I've edited the .bvr file and found out that if I remove the quotes from the TB entries all works fine.

eg.
```
<<Pin>>
Part	TB	Pin	Name	X	Y	Layer	Net
"Q7960"	(T)	1	"1"	2.92300010e+00	2.72099996e+00	1	"1V05S0_SW"
"Q7960"	(T)	2	"2"	2.97300005e+00	2.72099996e+00	1	"1V05S0_SW"
"Q7960"	(T)	3	"3"	3.02300000e+00	2.72099996e+00	1	"1V05S0_SW"
"Q7960"	(T)	4	"4"	3.07299995e+00	2.72099996e+00	1	"1V05S0_TG"
"Q7960"	(T)	5	"5"	3.07299995e+00	2.91499996e+00	1	"PPBUSB_G3H"
"Q7960"	(T)	6	"6"	3.02300000e+00	2.91499996e+00	1	"PPBUSB_G3H"
"Q7960"	(T)	7	"7"	2.97300005e+00	2.91499996e+00	1	"PPBUSB_G3H"
"Q7960"	(T)	8	"8"	2.92300010e+00	2.91499996e+00	1	"PPBUSB_G3H"

"C7940"	(B)	1	"1"	2.99499989e+00	2.81699991e+00	2	"PPBUSB_G3H"
"C7940"	(B)	2	"2"	2.74499989e+00	2.81699991e+00	2	"GND"
```

corbolais, 2019-07-24:
@PiezPiedPy want to give https://github.com/OpenBoardView/OpenBoardView/pull/64 a try?

morganchristiansson, 2023-01-28:
> I've edited the .bvr file and found out that if I remove the quotes from the TB entries all works fine.

Thanks - I can now see both sides of the board.

davidebeatrici, 2023-06-04:
In addition to that, the delimiter should be specified as follows:

```
--delimiter='\t'
```