OpenBoardView/OpenBoardView #37: TOPTEST BoardViewR4 file format (.brd)

piernov, 2016-07-30:
Not an issue as it is done, just for keeping track of the supported formats.

inflex and I recently implemented another format which holds the extension ".brd" but is different from the one already supported. This format is plain text (not encoded). We called it BRD2 as we had no better idea, but it's not related in any way to the BRD format.

Implementation is available on inflex's branch:
https://github.com/inflex/OpenBoardView/commits/inflex-ui-features

Also implemented is format recognition based on common patterns rather than on the extension (to avoid conflict between the two ".brd" formats).

Attached is the binary and a boardview file.

[BoardViewR4.zip](https://github.com/OpenBoardView/OpenBoardView/files/391907/BoardViewR4.zip)

lerocm, 2016-09-29:
Avira is flagging the .exe inside BoardViewR4.zip as TR/Patched.gen.
Have attached a "clean" version  - well Avira thinks its ok at least.
[image]
[BoardViewR4.zip](https://github.com/OpenBoardView/OpenBoardView/files/500497/BoardViewR4.zip)

piernov, 2016-09-29:
VirusTotal analysis

Mine:
https://www.virustotal.com/en/file/58496c329d277a9eabfa9565b31acb09984cdec6e4db282f906bbbca0185c4f9/analysis/1475148495/

Yours:
https://www.virustotal.com/en/file/98ffc2be591c3ddc54e9adc21474c311b65f314c7e7be21421d1a984de25ace9/analysis/1475148574/

Both look suspicious but it might be false positives…

lerocm, 2016-09-29:
> VirusTotal analysis
>
> Mine:
> https://www.virustotal.com/en/file/58496c329d277a9eabfa9565b31acb
> 09984cdec6e4db282f906bbbca0185c4f9/analysis/1475148495/
>
> Yours:
> https://www.virustotal.com/en/file/98ffc2be591c3ddc54e9adc21474c3
> 11b65f314c7e7be21421d1a984de25ace9/analysis/1475148574/
>
> Both look suspicious but it might be false positives…

Thats annoying. Hope they are false positives. Have attached another one
that does a little better (add exe suffix).

https://www.virustotal.com/en/file/6810f759362779dd00dec02165437a457a9f0a40047fc73c573cf4a99021f10a/analysis/1475153060/

inflex, 2016-09-29:
I ran in to this a lot before, all false positives.  It is very annoying, a lot is a result of some looser AV engines picking up on too many things.  I've not yet encountered a false positive from the likes of ESET NOD32 or other similar reputable AV engines.  

The other thing is, if you change the optimisation or compiler options, you'll get different outcomes on the false positives.

MattSturgeon, 2016-10-01:
@lerocm  While I appreciate you taking the time to report this, it has nothing to do with the issue being discussed on this page; "TOPTEST BoardViewR4 file format (.brd)". I will remove the virus false positive discussion to keep this issue clean, however please feel free to create a new issue to track this.

In future please use the "New Issue" button, unless you believe your issue is already reported (in which case commenting is absolutely fine).

Thanks!

inflex, 2016-11-02:
This file format is supported in the merged OBV (R7.3 onwards).  Closing.

closed 2016-11-02 by inflex