OpenBoardView/OpenBoardView #38: Include some test files

tommis, 2016-08-13:
Some people might just want to test out this software so it would wise to include some free examples to try out. Arduinos have schematics available under Creative Commons Attribution Share-Alike license so they're okay to include.

https://www.arduino.cc/en/uploads/Main/arduino_Uno_Rev3-02-TH.zip

piernov, 2016-08-13:
The `arduino_Uno_Rev3-02-TH.brd` file is an Eagle file which is not supported by this software (and will probably never be since it's not its purpose). Probably all boardview files supported by this software are under copyright (and confidential too). We could however make a boardview for a simple (free) design and add it to the repository. @inflex may have some idea on this.

inflex, 2016-08-14:
Thanks - yes, I'll conjure up a sample board at some point.