OpenBoardView/OpenBoardView #40: Flip board, maintain position

maxsnts, 2016-08-18:
Hello

Would it be possible to: When flipping the board with the Space bar, position the board so that the point under the mouse match on both sides?

So we could position the mouse over the board, press space bar, and immediately see whats directly behind.

Thanks

inflex, 2016-08-18:
Do you think we should still have an option to do the current flip mode?  Perhaps bind with shift to modify behaviour?

inflex, 2016-08-19:
@maxsnts I've implemented this in the latest commit of my fork if you want to try it out to see if it's worth implementing here.

29453061c09b01c548a41684f7c2bb28a0003ed3

maxsnts, 2016-08-19:
To be honest i don't get the reason behind the "old" method of flip, and a think this one should be default. 
Was it just a "random choice" or is there a useful reason for the "old" filp?

inflex, 2016-08-19:
Personally I've infrequently needed the "what's on the other side at this point" flip, usually i'm flipping because I'm looking around in general as I've not yet found my desired area of interest.

I prefer to keep the more specific behaviour of the point-flip as the shift+flip as it produces a more pronounced effect and is dependent on the current mouse position.

Later on when we get to programmable key-codes for actions that's obviously something that can be adjusted by everyone (for now, you can simply adjust the source code, it'd only take the inclusion of a single ! character; ie, ` if (!io.KeyShift)  {....`

MattSturgeon, 2016-08-19:
Maybe we should have a modifier to flip horizontally too...

Planning to get this merged in along with the reset zoom/position and some of your movement method abstractions At Some Point™, though feel free to make a PR in the mean time.

Not sure which flip behaviour should be default though.

maxsnts, 2016-08-19:
Its probably a matter of opinion/preference.
My 2 cent, the "new" flip method allow us to see what directly behind, if we don't need that it still works.
I think you could skip the modifier entirely (one less to remember and press). 
The mouse could act as the modifier: if its outside of the board, if flips "the old way" if its on the board it flips "the new way".  
Thank you for implementing it whatever way you choose.

inflex, 2016-08-20:
@maxsnts I've added a selectable flipMode in to the File->Preferences  panel now, so you can select which ever is preferred (and the other mode will be activated using shift).

60fd1c7027a7dc806bd83270c411b19ac0c29c31

maxsnts, 2016-09-25:
Oh man... the "mouse flip" together with that "connection trace" is just amazing.
I can click a pin or net, see where the trace leads, middle click to flip and immediately see where it connects.
Thank you.

inflex, 2016-09-25:
You're welcome @maxsnts .   Did you see R7.1 was put out a few hours ago.  Two minor tweaks to R7 mostly to do with the zoom level when you select nets/parts and search results.

Thanks for bringing the feature to mind, I must admit, now I use it more than I realise, thankyou.

maxsnts, 2016-09-25:
I have a confection to make... while i'm a professional programmer, i find my self not wanting to do code on my time. Electronics is my "first love" and i only get to do it on my leisure time.
I realize that with my background, i should be helping instead of "requesting", but i sorry, i lack the will to dive into the code since i spend all week doing it.
I really appreciate the time to place into this. 
Many thanks.

inflex, 2016-09-25:
@maxsnts that is quite okay.  There are other projects in the world where I ask for features but don't code them myself; we cannot do every project.  For me, electronics and programming are my job, and since I work for myself, and improving the software helps me finish jobs faster, it's only natural that I would put the time in to this project.

A lot has changed in Openflex/Openboardview since you originally requested the feature.    Hopefully soon we'll merge Openflex back in to Openboardview and carry on as a single branch again.

maxsnts, 2016-09-25:
If i could be so bold, not wanting to take advantage of your time more than i already have.
A question outside of the scope of this project, would that be OK?
(Its ok if you say no, it really is)

inflex, 2016-09-25:
May as well ask,  though if you prefer, you can always ask outside of this domain, you can contact me at pldaniels [at] gmail dot com

closed 2016-09-25 by inflex