OpenBoardView/OpenBoardView #41: Build error - concatenating mismatched

mrxordi, 2016-08-23:
Bulid error
1>C:\Users\XoRdi\Desktop\OpenBoardView\src\main.cpp(216): error C2308: concatenating mismatched strings
1>          Concatenating wide "%s - " with narrow "Open Board View"

When building MSVC2013
Add on the top of main.,cpp
`#define W_(val) L ## val
#define W(val) W_(val)`
And change line 216 to:
`			_stprintf_s(title, _countof(title), L"%s - " W(PRETTY_NAME), filename);`

piernov, 2016-08-23:
I think we did not have any issue with VS2015. Did something change between VS2013 ans VS2015 ?

inflex, 2016-08-23:
VS2015 seems to have been a *significant* jump.  The common runtime is all very different too.  

If you're building this codebase I'd strongly suggest doing so with VS2015.

mrxordi, 2016-08-23:
All i found is this:
http://stackoverflow.com/questions/2192416/how-to-convert-concatenated-strings-to-wide-char-with-the-c-preprocessor

piernov, 2016-10-30:
Not relevant anymore since the switch to inflex-ui-features.

closed 2016-10-30 by piernov