OpenBoardView/OpenBoardView #42: GetFontData in some cases returns GDI_ERROR

mrxordi, 2016-08-24:
platform_win32.cpp - load_font
In case(my case) of error GetFontData returns GDI_ERROR which is
#define GDI_ERROR (0xFFFFFFFFL)
which then throws exception when creating the buffer, we need to check against that ocasion

piernov, 2016-08-24:
Thanks, do you have any idea why it would fail?

mrxordi, 2016-08-24:
I'll do some research, however program runs fine for me after that check, even when it fails.

mrxordi, 2016-08-24:
Okay, it looks like CreateFont creates handle even when the font isn't present in the system. I don't know why it returns valid HFONT handle, what failed after creating non-existent font was GetFontData.  I didn't have "Liberation Sans" and "DejaVu Sans" fonts, when i installed them it stopped.

It doesn't change the fact that sanity check must be there, another thing is maybe you should enumerate the fonts by EnumFontFamilies first.

MattSturgeon, 2016-08-24:
Should be fixed in d53c526759a6063ee613c0409ed02e9fd2277902, thanks :smile:

closed 2016-08-24 by MattSturgeon