OpenBoardView/OpenBoardView #45: Asus TSICT file format (.asc)

piernov, 2016-09-04:
Not an issue, just for keeping track of the supported formats.

I just implemented support for the `*asc` files. This format is plain text (not encoded). This format relies on a directory containing the following files: `format.asc`, `parts.asc`, `pins.asc`, `nails.asc`, `nets.asc` and sometimes a `.bom` file.
It is very similar with the `.bdv` format since in fact, the `.bdv` format is made with concatenated `format.asc`, `pins.asc` and `nails.asc` and encoded.
Basic implementation is available on my [`feat/asc-support`](https://github.com/piernov/OpenBoardView/commits/feat/asc-support) branch and will be merged in `inflex-ui-features`. See inflex/OpenBoardView#43 and inflex/OpenBoardView#55 .
It supports loading `format.asc`, `pins.asc` and `nails.asc` only.

Attached is the binary and boardview files.

[Asus F5V, r1.1.zip](https://github.com/OpenBoardView/OpenBoardView/files/454039/Asus.F5V.r1.1.zip)
[X550CCR2.zip](https://github.com/inflex/OpenBoardView/files/416996/X550CCR2.zip)
[ASUS TSIC.zip](https://github.com/OpenBoardView/OpenBoardView/files/454040/ASUS.TSIC.zip)

inflex, 2016-11-02:
Available now in the merged OBV R7.3 onwards.

Do we want to close this @piernov  ?

piernov, 2016-11-02:
We don't support all the information available in those files, but we can add them later as needed. I left the issue open to "document" about the supported formats. We should really write some pages in the Wiki (or elsewhere) about them…

MattSturgeon, 2016-11-02:
@piernov the README is a good place to list supported formats, if you want to go into more detail then the wiki makes sense

Regarding keeping"probably fixed" issues or similar open: my preference is to close the issue when you think it is fixed or finished to a "good enough" standard and then open new, smaller issues if needed in the future.