OpenBoardView/OpenBoardView #46: exe is being flaged by anti-virus software

MattSturgeon, 2016-10-01:
> *From @lerocm*
>
> Avira is flagging the .exe inside BoardViewR4.zip as TR/Patched.gen.
> Have attached a "clean" version  - well Avira thinks its ok at least.
> [image]
> [BoardViewR4.zip](https://github.com/OpenBoardView/OpenBoardView/files/500497/BoardViewR4.zip)

inflex, 2016-10-01:
This sort of thing is going to pop up frequently as we progress towards a more stable build. Unfortunately in order to try 'catch everything' a lot of the less sophisticated AV engines get a bit loose on their heuristics and subsequently have a greater number of false positives compared to more advanced engines such as ESET NOD32 (interestingly, a lot of them use the same signature databases but implement different scanning engines).

I noticed this is happening a lot less now that we're using the linux->windows cross build / travis.

ie, latest R7.2 results - https://www.virustotal.com/en/url/7e4448c902158b86ad9fea5bbd2069d1ff93bc384ed55b2d1fe397693ecc43f7/analysis/1475305558/

MattSturgeon, 2016-10-01:
Intersting. Linux tools are better at making windows binaries than windows
tools! :D

inflex, 2016-10-01:
I suspect it's the different included libs / codebase cutting down on the level of commonly appearing sequences. Not sure, it could just be coincidence, but I know the first few VisualStudio2015 builds were hit with up to 4 or 5 false hits each time.

piernov, 2016-10-30:
Not relevant, link will be removed anyway from #37

closed 2016-10-30 by piernov