OpenBoardView/OpenBoardView #47: Sleep

haakonnessjoen, 2016-10-13:
I just compiled the unix branch on my mac. And while it starts up. It uses all the CPU on my mac. I look at main.cpp, and I see a mainloop without any sleep.

You should add a sleep here, of f.ex 20ms. There is no reason for this app to try to have 2000FPS.

piernov, 2016-10-13:
You may want to check out @inflex fork at https://github.com/inflex/OpenBoardView , it does have all that's needed regarding your issue.
We have plans to merge back @inflex work in this official repo but there is still no progress.

haakonnessjoen, 2016-10-13:
I just added a usleep() myself. And the cpu was happy.

The other one was harder to get to compile, I had to trick it to know where SDL2 was located.
Seems like the GUI is changed a lot too. Not as slick as your version. Too large fonts, etc.

inflex, 2016-10-14:
Font size is changable in the other fork as well from the UI.

I'm guessing you're trying to build a Windows version, because we now build the Win fork version using a linux cross compile to avoid the issues with VisualStudio.

piernov, 2016-10-30:
Fixed as of R7.3

https://github.com/OpenBoardView/OpenBoardView/releases/tag/R7.3

closed 2016-10-30 by piernov