OpenBoardView/OpenBoardView #50: Rendering issue on Linux

dfear, 2016-10-27:
The first time I load a board file in the OpenBoardView (OBV), the board is rendered as shown in the attached screenshot. 

> I prefer the dark base theme, so thats what I set in colour preferences.

Every time I open OBV and check colour preferences, the base theme is always set to light theme, even when I have previous set it to dark theme, clicked done, then restarted OBV.

When I load a board file for the first time it is rendered in neither a light theme nor a dark theme, it seems somewhere in between.

[image]

I am running OBV Build R573 /tags/R7.2 on Ubuntu Gnome 15.10 using FGLRX (drivers) 2:15.201-0ubuntu2-15.10.2.

inflex, 2016-10-27:
I've seen this before when i've been working on new things - trying to remember what causes it.

Best I can suggest is just remove the OBV configuration file in ~/.config/openboardview/obv.conf and run the program again.

inflex, 2016-10-27:
Else try attach your obv.conf file here so I can see what's happening.

dfear, 2016-10-27:
[obv.conf.zip](https://github.com/OpenBoardView/OpenBoardView/files/556789/obv.conf.zip)

dfear, 2016-10-27:
@inflex - i have removed the obv.conf file and that solved the problem! 

Thanks for the help.

I have attached my obv.conf file so you can see what's happening.  
[obv.conf.zip](https://github.com/OpenBoardView/OpenBoardView/files/556789/obv.conf.zip)

closed 2016-10-31 by inflex