OpenBoardView/OpenBoardView #53: 0.40 CRITICAL issue. Crash on startup - 0.3 works fine

psycodeoday, 2016-10-30:
after launch OpenBoardView.exe it crash suddenly (you cannot see even the main work window)
both zip/portale version and msi installer crash.
but in the same time 0.3 version (and older) works fine.

my setup:
i7, 8gb ram, gtx 980
win 7 sp1 64bit, clean, net-framework 4.6.2, java 8, eset smart security 8, fresh nvidia drivers, direct x 9.0c

[image]

[image]

piernov, 2016-10-30:
Try with https://github.com/piernov/OpenBoardView/releases/tag/N12

piernov, 2016-10-30:
Should work with R7.3, please try it:

https://github.com/OpenBoardView/OpenBoardView/releases/tag/R7.3

closed 2016-10-30 by piernov

psycodeoday, 2016-10-30:
both versions works fine :)
thank you :)