OpenBoardView/OpenBoardView #56: Error opening a Board View.

eduhalts, 2016-11-01:
Hi, I´ve just downloaded your software but I get an error when I tried to open a .brd file which I can open with Landrex Tlink drawing program...  I´m attaching the file for testing.
Thanks in advance.

[image]

piernov, 2016-11-01:
Probably invalid file header or truncated file. You didn't actually upload it though.

inflex, 2016-11-01:
I'll sort out a more graceful/meaningful error system, but it'd definitely be good to get our hands on a copy of that board file that's causing the failure.

eduhalts, 2016-11-01:
How can I send it to you?  I couldn´t attach it here.  It wasn´t a valid file type for attaching here.  I´m not sure if it is corrupt because I can read it wint another Board View reader.

inflex, 2016-11-01:
You can email me a copy to repairs@ctpc.biz  else zip and attach it, that should work.

@piernov and I were talking about it earlier, likely the other software also notices the errors in the files but it just proceeds to display it regardless.

eduhalts, 2016-11-01:
I´ve just sent it.  Hope it helps.

inflex, 2016-11-01:
@piernov here's the raw dump and the BRD in zip format.  Seems the input decoder is thinking there's only 26 nails requested (but I think it should be 2694?)
[brd-broken.zip](https://github.com/OpenBoardView/OpenBoardView/files/564201/brd-broken.zip)

[broken-brd-raw.txt](https://github.com/OpenBoardView/OpenBoardView/files/564197/broken-brd-raw.txt)

inflex, 2016-11-01:
Found the fault... the unit/section tests are assuming specific case of lettering;

ie, "Format:"  but the BRDFile is searching for "format:", so it never changes decoding mode.  Will have to make it case-insensitive search.

eduhalts, 2016-11-01:
There is something I can do to open the file?

inflex, 2016-11-01:
Not yet.  Hopefully will have it sorted out by tomorrow. While the changes in the section delimiter names is easy enough to deal with, there are some other aspects which have been encoded differently as well.

eduhalts, 2016-11-01:
I hope so.  This other program is driving me crazy!  I´m just learning to do this, and the other program doesn´t help.  Thanks in advance!

inflex, 2016-11-01:
@piernov this really is a different "BRD" format again; couple of things already;

- different section labels
- end_of_pins is actually *start* of pins 
- there's no net name to read in the Pin2: data, that has to be pulled from Nails:
  
As for U26 being odd, probably some sort of 1-off indexing issue.  Trying to first see if I can map the net names to the pins.

inflex, 2016-11-01:
Partially working now;  unfortunately I don't have a reference for this board to know what the pin nets *should* be so I can determine if I'm walking in the right direction or not.

[image]

eduhalts, 2016-11-01:
I´m goint to send the Schematics to you.

inflex, 2016-11-01:
Okay, I think we're about on the home run now (so much for sleeping :( ).

Notice how U26 isn't messed up now, and the net names aren't strange.

[image]

inflex, 2016-11-01:
I don't have the binary other than for linux,  but this should resolve it.  I would prefer that it's checked over by @piernov before I do a PR to master.

https://github.com/inflex/OpenBoardView/tree/R74Dev

inflex, 2016-11-02:
@eduhalts try one of the binaries from this build I did for you 
   https://github.com/inflex/OpenBoardView/releases/tag/R74Pre1

eduhalts, 2016-11-02:
Now opens without errors!
I wonder how to do to put the dark bacground on the board.  The only problem now seems to be with the messages in the bottom line.
[image]

inflex, 2016-11-02:
The  overlap  of the text at the bottom will be due to just running out of space.  I'll work on that in the future to make it not overrun the status details.

If you want the dark theme, just got File->Colour Preferences->Dark theme

eduhalts, 2016-11-02:
I think something is still wrong.  I was looking for pin 19(-pwrswitch), in Keyboard Switch (J7).  At first it showed the J7 part fine, I made an annotation, but now it is showed as 
[image]
J22, and pin 19 (and a lot of them) appears as Unconnected.

eduhalts, 2016-11-02:
I´ve just exited from the program and selected again the same board view.  Now it shows fine the bottom line, the J7, and the pwrswitch pin.  When I selected one of the view options, it scrambled the bottom line again.  It is repeating the top line (file name) over the bottom line.  After I tryed to exit again, now it scrambled the bottom line when I select the file to open.
Best regards.

eduhalts, 2016-11-02:
I´m sorry.  Reggarding the J7 vs J22 issue, was my mistake.  I was looking the bottom of the board.  I got confused because it showed my annotation over it.

inflex, 2016-11-02:
The annotation should only be showing on the side of the board where it was placed.   If it was originally created for J7 then it shouldn't be showing up on J22.

eduhalts, 2016-11-02:
You´re right.  I had annotations in both,  I don´t remember if I wrote it again in J22 by mistake.  I think I need some sleep.  Sorry about that.  It seems that the only problem is the bottom line repeating the top line mixed with its data.

inflex, 2016-11-02:
I've already resolve the text overlap.  It looks more messed up because of the positional information rapidly changing and varying in size.  The new version puts a nice gap between; though I may yet improve on that and make it right-align the text readout so that we always see the filename as the priority rather than being pushed off the side, though it's a minor item to deal with and put in to R7.4 later.

eduhalts, 2016-11-02:
Do you think you have to repeat the file name in the bottom line, if it is already showed in the top line?

inflex, 2016-11-02:
In linux, there are some WMs which won't leave enough room in the title bar, or they use frameless windows, or if you're in full-screen mode; as such I prefer to keep the filename listed down the bottom for now as well as the title bar.

eduhalts, 2016-11-02:
Ok, that´s fine.  I really like your software.  Thanks again for your help.

inflex, 2016-11-02:
I'll mark this as closed.  There's a few other UI cleanups I need to make before R7.4 so I'll open a separate thread for that.  The primary issue of this thread has been now resolve (barring someone finding a bug in the decoder ;) )

closed 2016-11-02 by inflex

eduhalts, 2016-11-02:
Thanks again!

eduhalts, 2016-11-26:
Hi again.  I have this other file which I cannot open with openboardview.
I can open it with this one.
*http://notebookschematic.com/download/BoardViewR4.exe
<http://notebookschematic.com/download/BoardViewR4.exe>*

Best Regards.