OpenBoardView/OpenBoardView #58: Please include .desktop and extensions file

srakitnican, 2016-11-13:
So this two files are required for packaging. In Fedora at least, GUI applications that are packaged in official repository must have this. I think it makes sense shipping them with sources as it can be reused for other distributions as well. I've attached something I came up with as an example or as is.

openboardview.desktop
```
[Desktop Entry]
Encoding=UTF-8
Name=OpenBoardView
Comment=Software for viewing .brd files
Exec=openboardview -i %f
Icon=openboardview
MimeType=application/x-openboardview-board
Terminal=false
Type=Application
Categories=Utility;Viewer;Electronics;GTK;
```
openboardview.xml
```
<?xml version="1.0" encoding="utf-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-openboardview-board">
    <sub-class-of type="text/plain"/>
    <comment>OpenBoardView PCB Layout</comment>
    <glob pattern="*.asc"/>
    <glob pattern="*.bdv"/>
    <glob pattern="*.brd"/>
    <glob pattern="*.bvr"/>
    <glob pattern="*.fz"/>
  </mime-type>
</mime-info>
```

MattSturgeon, 2016-11-13:
IIRC I had these files generated by the CMake scripts before merging back inflex's changes. I'm on my phone at the moment, but I'm sure someone can look at getting these files included in the build

srakitnican, 2017-07-18:
[openboardview.appdata.zip](https://github.com/OpenBoardView/OpenBoardView/files/1156900/openboardview.appdata.zip)

```xml
<?xml version="1.0" encoding="UTF-8"?>
<component type="desktop">
  <id>openboardview.desktop</id>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>MIT</project_license>
  <name>OpenBoardView</name>
  <summary>Viewer for PCB layouts</summary>
  <description>
    <p>
      Software for viewing PCB/Laptop/Motherboard Layouts.
    </p>
    <ul>
      <li>Dynamic part outline rendering, including complex connectors</li>
      <li>Annotations, for leaving notes about parts, nets, pins or location
      </li>
      <li>Configurable colour themes</li>
      <li>Configurable DPI to facilitate usage on 4K monitors</li>
      <li>Tablet usage with OSD controls</li>
      <li>Slower CPU systems through adjustable features</li>
      <li>Reads FZ (with key), BRD, BRD2, BDV and BV* formats</li>
      <li>Combined search system finds parts, pins and nets all within single
          search</li>
    </ul>
  </description>
  <screenshots>
    <screenshot type="default">
      <image>http://openboardview.org/ofbv-light-dark.png</image>
    </screenshot>
  </screenshots>
  <url type="homepage">http://openboardview.org</url>
</component>
```

MattSturgeon, 2017-07-18:
If you look in, for example, the old [`unix` branch](/OpenBoardView/OpenBoardView/tree/unix/asset)  you'll see that before we were able to have CMake templates for desktop files (etc) and have them built using our CMake variables (such as project name, version, etc).

Unfortunately this was lost when we reconciled the forked versions a while back, but I'm sure one of the current devs would be willing to take a look at the old setup and try and replicate it so that distros can use our files. Otherwise, feel free to do so yourself and send a PR.

I'm sorry that I've not been able to look into this myself.

MattSturgeon, 2017-07-18:
@OpenBoardView/maintainers

srakitnican, 2017-07-20:
Thanks for the pointers. I might do a research on how to do that if nobody takes it.

srakitnican, 2017-07-20:
I was thinking actually of just adding the files as is somewhere in the tree (asset probably) after an agreement on the format. Does it really need to be generated, this stuff most of the stuff, like name would not change that often, if at all. In that case it could be just changed with a git commit.

MattSturgeon, 2017-07-20:
If I had my way all project strings would be set in some central place and
then used as variables templates.

You could even have a header file template to include the variables into
the source code.

It just makes it easier to manage when strings aren't duplicated all over
the place, especially version numbers.

I might have 5 mins to look into this, after a quick coffee, but don't hold
me to it.

piernov, 2017-07-20:
I don't think it needs to be generated in this case. (I would even prefer it not to be, but that's personal taste.)

You can add those to the `asset/` folder. It would have to be installed in the correct destination by CMake for the RPM and DEB packages.

srakitnican, 2017-07-20:
Added a pull request just for files, for CMake I currently have no idea how to add it. I've studied freedesktop specifications and this is what I come up with:

asset/ofbv-app.svg --> /usr/share/icons/hicolor/scalable/apps/openboardview.svg
asset/openboardview.appdata.xml --> /usr/share/appdata
asset/openboardview.desktop --> /usr/share/applications
asset/openboardview.xml --> /usr/share/mime/packages

In rpm spec file I am currently installing them manually since make install needs a lot of work.
https://copr-be.cloud.fedoraproject.org/results/srakitnican/openboardview/fedora-26-x86_64/00581077-openboardview/openboardview.spec

piernov, 2017-07-20:
We're using CMake with CPack to automatically generate RPM and DEB packages, both should contain those files too. The `asset/CMakeLists.txt` will need to be updated for that purpose (though I don't have an idea how right now).

Also the `sub-class-of` in the MIME info file is not quite right: only `.asc` and `.bvr` are actually plain text files, the others are encoded/encrypted if I remember fine.

For the specific case of `.brd`, there are multiple formats with this extension, we only support 2 of those. Using a `magic` field in this case would be better, but that's not necessary at all.

srakitnican, 2017-07-21:
Just pushed commit 422c2f9, better?

srakitnican, 2017-07-21:
For the inclusion into CMake, it seems we need to just add the files to regular "make install" and they will be included into packages as well.
https://cmake.org/cmake/help/v3.6/module/CPack.html

MattSturgeon, 2017-07-24:
Fixed in befc555b3dc32b2549eef3c9e7a33ca2c226e86c

closed 2017-07-24 by MattSturgeon