OpenBoardView/OpenBoardView #59: Fileformat: CAST ( .LST plain text / .CST 8-bit )

inflex, 2016-11-15:
File format to implement, at least this one looks moderately rudimentary being plain text (the .LST parts at least, the CST seems 8-bit)

This package contains a viewer as well, and a pdf

[cast.zip](https://github.com/OpenBoardView/OpenBoardView/files/591816/cast.zip)

piernov, 2016-11-15:
[image]
Nowhere near being usable but it's a start.

inflex, 2016-11-15:
Less than 12 hours and you're already at that stage, I'm impressed!

What's left still to do?

piernov, 2016-11-15:
For the most basic support, I have to figure out which field tells on which side the component is (should not be too hard) and I have to parse the nets list and find the corresponding field in the pins (should be rather easy too). Also the current code is a dirty mess. I still have to find out where the board outline is defined. Then come the test points.
This format also packs in a few extra features not supported in OBV like vias, traces and pin/component shapes.
Like FZ it provides extra information on the parts in the LST file which is not parsed at all for now.

inflex, 2016-11-15:
A shapes library, with standard outlines is certainly something that needs to be worked on in the future, though at least for now the rendering is "sufficiently ample" for the task at hand, but it's always nice to do better when there's nothing else left to do :+1: 

The part information at least we can currently display like we do with the FZ format.  While it's not separated/parsed in to smaller units (name/value/outline) at least the collective information can still be shown [in the side bar or hover]

piernov, 2016-11-16:
[image]

Looks better. The side of the board is defined by the layer, I need to figure out if there's way to know the number of layers, for now 1 & 12 are hardcoded.

inflex, 2016-11-16:
Can you process using 1 = top (or bottom), anything > 1 = other side?  Or wait till you know the maximum number of layers and then post-process the data to set the side accordingly?

piernov, 2016-11-16:
Committed my work here: https://github.com/piernov/OpenBoardView/commit/0b8ce0f394751e2fca0463a991c69b27eee7dd8b
Not finished but I'm not sure I will have the time to do it…

inflex, 2016-11-23:
Do you think it might be worth pushing that to master?  The testing I did worked fine.

lordelectrode, 2017-08-15:
Just gonna link relevant fixme for hardcoded layer 12: https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/FileFormats/CSTFile.cpp#L94

piernov, 2017-08-15:
I guess @inflex suggestion is relevant, I can't find where the total number of layers would be defined.

lordelectrode, 2017-08-16:
`if (layer > 0x01)` seems to work

inflex, 2017-08-16:
I'd be more inclined to set it to > 5 or 6 probably, as it's entirely possible that we may run in to future boardview files that perhaps use 1..4 perhaps as top layers.   It's a guess only but I think it'd be prudent to leave a little room for other potential "top layers".

piernov, 2017-08-16:
I assume those are only electrical layers. It would be weird to put components on the inner layers…

inflex, 2017-08-16:
I agree, it would be strange to do that @piernov , but not to say It wouldn't happen.  Maybe they export-mapped to 1 for outline and 2 for top. Not sure, but just thought at least if there was any data in there, it'd still be mapped as a failsafe.  Or we could just dump a warning.