OpenBoardView/OpenBoardView #60: Fileformat: PCB (PADS)

inflex, 2016-11-15:
For a lot of people who've been trying to get iPhone boardviews, i've noticed there's now  the .pcb files around which look somewhat like a database format (all the field values are blocked together).

There is a viewer available, if you google for "PADS PCB viewer"

[sample-pcb.zip](https://github.com/OpenBoardView/OpenBoardView/files/591819/sample-pcb.zip)

RavetcoFX, 2017-09-07:
Here is the iPhone 6 Plus boardview in asc format if that helps, it still won't open with the latest version though :/

[iPhone6Plus-5.5.asc.zip](https://github.com/OpenBoardView/OpenBoardView/files/1283223/iPhone6Plus-5.5.asc.zip)

inflex, 2017-10-06:
[image]
We might finally have a break here.  Someone has created a simple binary file for us.

[pads_simple.zip](https://github.com/OpenBoardView/OpenBoardView/files/1361501/pads_simple.zip)

RavetcoFX, 2017-10-10:
@inflex is that screenshot from openboardview?

inflex, 2017-10-10:
@RavetcoFX not sure where that one is from, was supplied to me from the source that provided the smaller pcb file.

Slydder, 2018-03-13:
@inflex 

Hey Paul,

Anything further on this?

chuck

mefistotelis, 2021-12-13:
We should probably start working on format specification. Ie.
```
0000h     2bytes    always 00 FF
0002h     2bytes    PADS version number? File format version number? ie. 26 20 or 27 20
0004h     6bytes    zero-filled
...
SIZE-2Eh  4bytes   zero-filled
SIZE-2Ah  26bytes UUID in string form, no idea what it means
SIZE-4h   4bytes    size of the whole file minus 46 (2Eh), 46 seem to be footer length
```

Without specification, it isn't easy to just figure out file format.