OpenBoardView/OpenBoardView #62: Fileformat: Cadence BRD file

inflex, 2016-11-23:
Yet another BRD file.  This one is a binary dump but it's not encoded. Seems to contain  a lot of information.  
[candence.zip](https://github.com/OpenBoardView/OpenBoardView/files/608992/candence.zip)

ShadowsNight1000, 2018-10-01:
I've got another .BRD file that I've only been able to open Allegro Free Physical Viewer 16.6 if your interested

gsuberland, 2020-04-10:
The [Open Compute Project Olympus Dual-Xeon motherboard](http://files.opencompute.org/oc/public.php?service=files&t=e969672c57d6e17647adea54f2c3e5a7&download) (unzip, then extract the inner Motherboard_Board_File zip) also appears to be Cadence Allegro format, and does not open.

clayfreeman, 2020-05-05:
It's been a few months since I've tried to reverse engineer this format, but I thought I'd post my notes for posterity.

The Allegro viewer referenced by @ShadowsNight1000 seems to use memory-mapped I/O to read the file. I tried to trace the reads to the memory region set up by the application, but I couldn't glean too much information about what the application was doing aside from what appears to be a version check in the file's header.

The viewer then appears to iterate over part labels for resistors, capacitors, etc, but I couldn't see where any placement information was being held.

Luckily, I don't believe there is any sort of embedded compression, encryption, or encoding of any kind aside from the container format itself; if you run `strings` on a file of this format, you get a readable list of part labels.

The reversing process would be much easier if there were multiple (as small as possible) test files available for figuring out the file structure (i.e. offsets and pointers) to notable things such as board dimensions, coordinates, units, etc. I'm not super familiar with electrical engineering or board design, so some help in this capacity would be useful for me as an outsider to this industry to further my investigation into this file format.

LorenzoAlfaro, 2020-05-18:
> It's been a few months since I've tried to reverse engineer this format, but I thought I'd post my notes for posterity.
>
> The Allegro viewer referenced by @ShadowsNight1000 seems to use memory-mapped [...]

> It's been a few months since I've tried to reverse engineer this format, but I thought I'd post my notes for posterity.
>
> The Allegro viewer referenced by @ShadowsNight1000 seems to use memory-mapped [...]

Hi clayfreeman, 

I am willing to be of assistance. I am also interested in reverse engineer the Cadence BRD format. But my background actually is more electronic than software oriented. I have some brd files that are much smaller and simple (almost bare bone boards). I uploaded one, let me know if you want to discuss this further.
[final_HDL.zip](https://github.com/OpenBoardView/OpenBoardView/files/4646297/final_HDL.zip)

jeffwheeler, 2024-04-08:
Hi, I also wanted to share my repository [brd_parser](https://github.com/jeffwheeler/brd_parser) that attempts to decode the binary Allegro `.brd` format. The API is not stable, but if you're interested in linking against it, I'm sure we could figure something out.

inflex, 2024-04-08:
> Hi, I also wanted to share my repository [brd_parser](https://github.com/jeffwheeler/brd_parser) 

Great to see that development.  I think the first issue at hand will be the licence mismatch ( MIT vs GPLv3 ),  beyond that we'd have to look at the Boost dependency perhaps.

Maybe a compromise solution would be creating a AllegroBRD -> GenCAD or similar exporting tool ( avoids licence issues and helps everyone still get access to the boardview );  we'd really only need
  * board outline
  * pins { ID/number, position, side, part name, net name }

swiftgeek, 2024-04-08:
Going with GenCAD could help avoid artwork copyright issues as well ([when stripped](https://github.com/swiftgeek/pcbtools/blob/main/gencad/gencad_strip_art.sh))

swiftgeek, 2024-04-08:
That would be a [reference designator](https://en.wikipedia.org/wiki/Reference_designator)

inflex, 2024-04-08:
:sigh:

inflex, 2024-04-08:
> That would be a [reference designator](https://en.wikipedia.org/wiki/Reference_designator)

I think it was fairly clear in the context of the general discussion that we're having here with boardview software that when I refer to "Part name" I'm referring to the "reference designator", given that the terminology is preexisting within the software itself.

No need to be that pedantic.

piernov, 2024-04-08:
@jeffwheeler  
> Hi, I also wanted to share my repository [brd_parser](https://github.com/jeffwheeler/brd_parser) that attempts to decode the binary Allegro `.brd` format. The API is not stable, but if you're interested in linking against it, I'm sure we could figure something out.

Thanks a lot for this work, that's awesome. Would it be possible for the lib part to be licensed under LGPL at the very least? This would allow OpenBoardView to use it as an external shared library. I'd still prefer a more permissive license (e.g., MIT), otherwise distributing e.g., Windows binaries becomes more challenging. (right now we have a single statically-linked executable)

jeffwheeler, 2024-04-09:
Hi all, great questions about the license. I selected GPLv3 because I was designing for integration into KiCad, which is also GPLv3. Actually, I've copied most of the code already into a KiCad branch. (I recognize @swiftgeek from the ticket over there where I was working on it, [#9239](https://gitlab.com/kicad/code/kicad/-/issues/9239). 🙂)

I'm not immediately sure what to think about licensing. I'll have to do some learning. With that said, I think you're welcome to look at my implementation and re-implement it. Most of the hard work is in figuring out the structure of the file, which you can find [here ](https://gitlab.com/jeffwheeler/kicad/-/blob/add_cadence_allegro_importer/pcbnew/pcb_io/allegro/allegro_structs.h?ref_type=heads) (KiCad branch) or [here](https://github.com/jeffwheeler/brd_parser/blob/main/src/lib/structure/types.h) (`brd_parser` repo). The gist of the file is a bunch of structs concatenated together after a header. The form of each struct is identified by the first byte. There's a fair bit of mess in both versions of the file that still needs to be cleaned up. You can reference the rest of the importer (KiCad branch) or GUI (`brd_parser` repo) to see how the structs are used.

Also, y'all mentioned GenCad, but I've never heard of that. I Googled around, but where's the best place to learn about that?

swiftgeek, 2024-04-09:
GenCAD is a legacy industry "de facto standard" for data exchange with eg. board asembly house. Its modern replacement is far more painful to work with and inspect, which is one of the reasons why GenCAD still remains popular (another being simply legacy setups involving legacy software of machines on assembly line).

Being pure ASCII, it's easy to inspect/troubleshoot/repair, which kinda comes handy when you have to usually work with CADs like cadence allegro or siemens (previously mentor) pads, and usually for those setups CAMCAD is used ([like here](https://www.parallel-systems.co.uk/wp-content/uploads/2020/02/Generate_gencad.pdf)).

From opensource tools, KiCAD can already generate usable GenCAD files ([like here with olimex s3 as example](https://github.com/OpenBoardView/OpenBoardView/pull/247)), though some issues/unexpected things will appear, that wouldn't have happened with CAMCAD generated GenCAD file.

OEMs also have certain conventions on how much data is shared in GenCAD files, most common one being leaving outer layers artwork, while removing inner ones.

Spec can be found eg. [here](https://web.archive.org/web/20240409031151/https://electronix.ru/forum/applications/core/interface/file/attachment.php?id=15091)

jeffwheeler, 2024-04-09:
Got it, thanks for explaining.

BinBashBanana, 2025-04-18:
@jeffwheeler your brd_parser repository was taken down yesterday, was this intentional?