OpenBoardView/OpenBoardView #65: Eagle board format?

obbardc, 2017-01-21:
Hi guys,

Looks like great software! This software would be really cool to aid assembly/testing of PCBs..

I (and a few others) use the Eagle XML format.  Any interest on adding in support for this?

inflex, 2017-01-21:
Working on it.  Just taking a bit of an extended holiday from OBV to recoup after a nearly non-stop 6 months of piling on the code.

At least the Eagle XML format is readable :+1:

SpeedD408, 2017-04-11:
This would be great.  I'm interested in using this software to help diag and repair old Amiga computers.

cpm1, 2017-11-22:
Just bumping as the latest version still doesn't seem to support Eagle BRD? For example: https://github.com/ChuckyGang/A3660/blob/master/A3660/A3660.brd

beliefsystemrecords, 2017-12-31:
would be immensely useful to be able to open eagle files!

eb0687, 2018-03-07:
I would be interested in this as well. Thanks.

inflex, 2019-05-24:
In the near future, there'll be added formats that Eagle can export to (Fabmaster, Hyperlynx) which you can then read in to OBV. Currently have been added to FlexBV and in refinement process.

nitrocaster, 2019-12-10:
I made a converter that takes Eagle BRD file and generates Toptest boardview file that OpenBoardView can read: https://github.com/nitrocaster/eagleview

spookydd, 2020-01-05:
> I made a converter that takes Eagle BRD file and generates Toptest boardview file that OpenBoardView can read: https://github.com/nitrocaster/eagleview

Which version of eagle can it go back to? I'm using an old eagle v6.1, will this work?
And how is this used anyway?

nitrocaster, 2020-01-05:
Usage is explained here: https://github.com/nitrocaster/eagleview/blob/master/src/eagleview.cpp#L12

Assuming your Eagle board file is `board.brd` and you want to make a boardview file `board_view.brd` from it, you should do this:

`eagleview board.brd -brd board_view.brd`

I didn't test it against board files created by various Eagle versions, but as far as I understand, the structure didn't change much - worst case scenario should require only minor tweaks. If you decide to try converting your boards and something goes wrong - please, let me know,

spookydd, 2020-01-05:
Ok, I figured it would be a simple CLI utility, so what I was wondering about isn't how it's invoked from the CLI but how to compile it. I'm on macos. There is no compiled version with a mac installer right?
I will test this on one of my boards, made with eagle v6.1, and we'll see how it goes.. But I need this compiled first.

nitrocaster, 2020-01-05:
OK, that's a good question that I can't answer right now. You're right, there's no binaries for Mac, as well as for Linux. I didn't use any Microsoft-specific language features or Windows headers, so it shouldn't be difficult to write a makefile and build it for your platform.

spookydd, 2020-01-05:
Cool. I'll be patient and wait for it then. I can test on mac any time.

QrchackOfficial, 2021-06-27:
> OK, that's a good question that I can't answer right now. You're right, there's no binaries for Mac, as well as for Linux. I didn't use any Microsoft-specific language features or Windows headers, so it shouldn't be difficult to write a makefile and build it for your platform.

I've just exported CMake to Unix Makefiles and managed to build on 10.14.6 Mojave (after enabling C++14 support and tweaking a few errors in the repo head), should anyone need a Mac binary, I seem to have gotten it working.