OpenBoardView/OpenBoardView #67: Bottom components show on top and vice-versa

phi7ip, 2017-02-23:
Hi I have attached 2 boardview files which have the same problem, some bottom components show up when viewing the top and some top components show up when viewing the bottom.
[LB# 820-2523-B.zip](https://github.com/OpenBoardView/OpenBoardView/files/795175/LB.820-2523-B.zip)

inflex, 2017-02-23:
Thanks for that +phi7ip , appreciate you putting the issue up after our discussion on the live stream.  I'm flat out today with work, so I'll try dig in to it tonight.

phi7ip, 2017-02-23:
It's the least I can do!

inflex, 2017-02-27:
In the file src/openboardview/Fileformats/BRDFile.cpp I added two lines directly after the part.type = READ_UINT() line.   Seems to have cleared it up

```
			case 4: { // Parts
				ENSURE(parts.size() < num_parts);
				BRDPart part;
				part.name        = READ_STR();
				part.type        = READ_UINT(); // Type and layer, actually.
				if (part.type == 1) part.type = 5;
				else if (part.type == 2) part.type = 10;

```
[image]

[image]

inflex, 2017-02-27:
Try this branch off my fork  https://github.com/inflex/OpenBoardView/tree/doublePartsFix

( @piernov is this the way you best recommend doing these sorts of fixes? )

piernov, 2017-02-27:
The various values part.type can take are handled here: https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/BRDBoard.cpp#L89

(the if - else if - else block selects the side, the following ternary operator selects the type)
Maybe it needs some adjustment there?

Btw it seems like you pushed the wrong stuff on your doublePartsFix branch.

inflex, 2017-02-27:
Thanks @piernov , not sure what kool-aid I've been drinking, I should have seen that.  I'll pull a fresh copy and try again.

inflex, 2017-02-27:
okay, how about this - 

https://github.com/OpenBoardView/OpenBoardView/tree/doublePartsFix

https://github.com/OpenBoardView/OpenBoardView/blob/doublePartsFix/src/openboardview/BRDBoard.cpp#L89

phi7ip, 2017-02-27:
wow that was fast, sorry to sound stupid but how do i build it on windows? i installed cmake but it still doesnt build with build.sh

piernov, 2017-02-27:
Building on Windows is not supported anymore. (maybe I'll fix it someday… been saying that for months) Windows binary is built by cross-compiling from Linux. @inflex can however make a pre-release and the binary would be automatically built&deployed from Travis.

inflex, 2017-02-28:
@phi7ip try the OBV binary listed here.  I cross-compiled it from my system locally as we seem to be having some issues with the autobuild engine right now on git.

https://github.com/OpenBoardView/OpenBoardView/releases/tag/PR20170228

phi7ip, 2017-02-28:
Great! Can confirm that they are where they're supposed to be now. Thank you!

I found a couple of other minor issues with the 820-2523 board

The container for some diagonal components is not at an angle, hard to explain but i have attached an image
[image]

And there seems to be something weird happening at the bottom 1/4 of the board
[image]

As i said these are only minor issues just thought i would let you know, great work so far!
Thank you @inflex  & @piernov

inflex, 2017-02-28:
@phi7ip you are correct, those parts aren't rendered the most appropriate way.  Part of the problem is that because we aren't supplied with an actual component package type the shape is subsequently determined using a minimum-bounding-box algorithm, and sometimes the minimum box is not what you were hoping for.   I have plans  for a further improvement on the algorithm that looks for the primary-axis of the pins to help bias the selection of the MBB.

On the 820-2523-B board, yes, part of the outline is missing, though it does seem to be missing content from the file itself, not broken rendering, at least that I'm aware of.  I'll double check this later.

phi7ip, 2017-03-01:
Yes that part of the board seems to be missing from the other readers too.

Thanks for all your work, it has also fixed the connectors appearing on the top side for all the brd files.

MattSturgeon, 2017-08-02:
I'm assuming this was fixed when we merged inflex's fork. Feel free to reopen if I'm wrong.

closed 2017-08-02 by MattSturgeon