OpenBoardView/OpenBoardView #70: Annotations can't be accessed on OSX

all2coolnick, 2017-03-16:
I have not tried this with a mouse but on a MacBook Pro 2012 running 10.12.3, I cannot access the annotations options as control clicking does not bring it up. I have tried all combinations of fn, ctrl, alt and cmd keys but can find no way of accessing this functionality.

inflex, 2017-03-16:
Thanks for the report - I'm presuming that's with the mac single button mouse?

all2coolnick, 2017-03-16:
Thanks for the reply.

This is on a laptop using the built in trackpad, not with a mouse, but control clicking on OSX should always simulate a right click on a mouse. I’ll try and get hold of an Apple mouse and double check it with that.

Thanks

Nick Gillard
The Bookyard Ltd.
Unit 18 Matchworks
(Banks Rd. entrance)
140 Speke Road
Liverpool L19 2RF

t. (uk) 08456 800 256
t. (int) +44(0) 151 427 4803
f. +44(0) 700 6040 566
w. www.thebookyard.com
w. www.mac2cash.com
e. nick@thebookyard.com

all2coolnick, 2017-03-16:
Just figured out that I can bring up the annotations box by clicking on an item with two fingers on the trackpad which is obscure but at least I know there is a way to do it now. 

Thanks

Nick Gillard
The Bookyard Ltd.
Unit 18 Matchworks
(Banks Rd. entrance)
140 Speke Road
Liverpool L19 2RF

t. (uk) 08456 800 256
t. (int) +44(0) 151 427 4803
f. +44(0) 700 6040 566
w. www.thebookyard.com
w. www.mac2cash.com
e. nick@thebookyard.com

all2coolnick, 2017-03-16:
FYI I tried it with the latest Apple Magic Mouse and right click does not open the annotations panel. I can use the two finger click but this is not documented anywhere, is not intuitive, and is contradictory to the current version of the user manual so I’m sure it will frustrate other users.

There also appears to be an odd omission which may be connected to the lack of right click functionality on MacOS. I can find no way of copying a component or net name which is pretty essential when moving backwards and forwards between the boardview and the schematic pdf. There appears to be no copy command and none of the places where the name of the selected object is displayed allows you to highlight it to copy. Is there a way to do this that I am not seeing?

Being able to copy a name from the schematic and then use the lighting fast search in openboardview to locate the matching item is fantastic but I need to be able to do the reverse and at the moment I have to re-type the full name of each item I search for in the pdf reader which is very slow for components or nets with long names such as “PP1V2_ENET_PHY_AVDDL".

Thanks

Nick Gillard
The Bookyard Ltd.
Unit 18 Matchworks
(Banks Rd. entrance)
140 Speke Road
Liverpool L19 2RF

t. (uk) 08456 800 256
t. (int) +44(0) 151 427 4803
f. +44(0) 700 6040 566
w. www.thebookyard.com
w. www.mac2cash.com
e. nick@thebookyard.com

inflex, 2017-03-16:
The lack of text cut-paste is something that's being worked on in the cross platform GUI toolkit (ImGui) that OBV is based on.  It's something that's been "pending" for a while though I've not seen any progress as yet.

As for the actual right-click emulation/behaviour, I'll have to look in to seeing if I can add a meta-key for macOS versions to facilitate that in a saner way.

inflex, 2017-03-16:
For reference, this is the ImGui thread pertaining to the issue - https://github.com/ocornut/imgui/issues/950

MattSturgeon, 2017-05-06:
It sounds strange that right-click works with two-finger click on a touchpad but not right-click on mouse, since two-finger click is considered a right-click by most input drivers (and three-finger click is middle-click) similar to two-finger scrolling and pinch zooming.

If Ctrl+Click is indeed considered a right-click by the input driver, it would be just as odd that it doesn't work, though it sounds more like something most applications choose to implement as opposed to a universal system-level thing.

ipatch, 2019-09-21:
please excuse me if i'm poking a sleeping bear, but I tested right click functionality on a 2013 late model MBP and can confirm <kbd>control</kbd>+<kbd>RMB</kbd> does not bring up the annotation menu.

however plugging or connecting a external mouse, (in my case a usb logitech mouse) and right clicking using the external mouse does indeed bring up the annotations menu.

I believe the reason @all2coolnick is able to bring up the annotation menu with the two finger tap / click on the trackpad is because he has a track pad that supports multi finger gestures and i believe most recent versions on macos have one finger on trackpad with the additional finger on track pad and clicking the trackpad will "simulate" a right / control click on macOS similar to how a RMB click on windows brings up a context menu.

Also, I've noticed another app that's built using GTK+ ie. Inkscape behaves similar where control clicking on the canvas does not bring up a "context" menu but using an external mouse does.  Not sure if was mentioned, but I'd like this this issue is related to GTK

cheers 🍩