OpenBoardView/OpenBoardView #74: Mingw_Runtime_Assertion

Cinek75, 2017-06-08:
Hello.
When trying to open a vinafix.vn_1000HE_MB_1.2G_boardview_encoded.fz file the openboard.exe (release R74Pre1)program crashes and reports this message 'Mingw Runtime Assertion...".
The screenshot and the file vinafix.vn_1000HE_MB_1.2G_boardview_encoded.fz I posted in the attachment.
Greeatings.
[image]
[vinafix.vn_1000HE_MB_1.2G_boardview_encoded.fz.zip](https://github.com/OpenBoardView/OpenBoardView/files/1060661/vinafix.vn_1000HE_MB_1.2G_boardview_encoded.fz.zip)

piernov, 2017-06-08:
Hi,
You're probably missing the FZ decription key. You can get it on the IRC channel #openboardview@irc.freenode.net and put it in the configuration file.

We should add an explanatory message rather than a wild assert to let the user know he has to fill the key in manually.

inflex, 2017-06-08:
Yes, agreed.  Another "small usability" item that needs to be cleaned up.  I've noticed I've run in to a couple of others lately that I need to resolve too (they're all simple, just need to be handled).

Cinek75, 2017-06-08:
Hello.
It is working now.
I manually copied the FZ keys to the end of file "AppData \ Roaming \ OpenBoardView \ obv.conf".
Regards.

closed 2017-06-08 by Cinek75