OpenBoardView/OpenBoardView #77: Support for GCC 4.8

srakitnican, 2017-07-18:
EPEL7 uses GCC 4.8, it would be nice if it would support it.

Tried to bypass checks and hit a wall:

```diff
cd /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview && /usr/bin/c++   -DENABLE_FONTCONFIG -DENABLE_GL1 -DENABLE_GL3 -DENABLE_GLES2 -DENABLE_GTK -DENABLE_SDL2 -DImDrawIdx="unsigned int" -DOBV_BUILD=\"7.3-3.el7.centos.x86_64\" -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches   -m64 -mtune=generic  -Wall -Wextra -std=c++11 -I/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview -I/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/.. -I/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/../imgui -I/usr/include/SDL2 -I/builddir/build/BUILD/OpenBoardView-R7.3/src/glad/include -I/usr/include/gtk-3.0 -I/usr/include/atk-1.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/pango-1.0 -I/usr/include/gio-unix-2.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/pixman-1 -I/usr/include/libpng15 -I/usr/include/libdrm    -pthread -o CMakeFiles/openboardview.dir/BRDBoard.cpp.o -c /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp
In file included from /usr/include/c++/4.8.2/algorithm:62:0,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/Board.h:6,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.h:3,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:1:
/usr/include/c++/4.8.2/bits/stl_algo.h: In instantiation of ‘_RandomAccessIterator std::__unguarded_partition(_RandomAccessIterator, _RandomAccessIterator, const _Tp&, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<std::shared_ptr<Component>*, std::vector<std::shared_ptr<Component>, std::allocator<std::shared_ptr<Component> > > >; _Tp = std::shared_ptr<Component>; _Compare = BRDBoard::BRDBoard(const BRDFile*)::__lambda1]’:
/usr/include/c++/4.8.2/bits/stl_algo.h:2296:78:   required from ‘_RandomAccessIterator std::__unguarded_partition_pivot(_RandomAccessIterator, _RandomAccessIterator, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<std::shared_ptr<Component>*, std::vector<std::shared_ptr<Component>, std::allocator<std::shared_ptr<Component> > > >; _Compare = BRDBoard::BRDBoard(const BRDFile*)::__lambda1]’
/usr/include/c++/4.8.2/bits/stl_algo.h:2337:62:   required from ‘void std::__introsort_loop(_RandomAccessIterator, _RandomAccessIterator, _Size, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<std::shared_ptr<Component>*, std::vector<std::shared_ptr<Component>, std::allocator<std::shared_ptr<Component> > > >; _Size = long int; _Compare = BRDBoard::BRDBoard(const BRDFile*)::__lambda1]’
/usr/include/c++/4.8.2/bits/stl_algo.h:5499:44:   required from ‘void std::sort(_RAIter, _RAIter, _Compare) [with _RAIter = __gnu_cxx::__normal_iterator<std::shared_ptr<Component>*, std::vector<std::shared_ptr<Component>, std::allocator<std::shared_ptr<Component> > > >; _Compare = BRDBoard::BRDBoard(const BRDFile*)::__lambda1]’
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:214:3:   required from here
/usr/include/c++/4.8.2/bits/stl_algo.h:2263:35: error: no match for call to ‘(BRDBoard::BRDBoard(const BRDFile*)::__lambda1) (std::shared_ptr<Component>&, const std::shared_ptr<Component>&)’
    while (__comp(*__first, __pivot))
                                   ^
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:212:46: note: candidates are:
  sort(begin(components_), end(components_), [](shared_ptr<Component> &lhs, shared_ptr<Component> &rhs) {
                                              ^
In file included from /usr/include/c++/4.8.2/algorithm:62:0,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/Board.h:6,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.h:3,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:1:
/usr/include/c++/4.8.2/bits/stl_algo.h:2263:35: note: bool (*)(std::shared_ptr<Component>&, std::shared_ptr<Component>&) <conversion>
    while (__comp(*__first, __pivot))
                                   ^
/usr/include/c++/4.8.2/bits/stl_algo.h:2263:35: note:   candidate expects 3 arguments, 3 provided
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:212:102: note: BRDBoard::BRDBoard(const BRDFile*)::__lambda1
  sort(begin(components_), end(components_), [](shared_ptr<Component> &lhs, shared_ptr<Component> &rhs) {
                                                                                                      ^
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:212:102: note:   no known conversion for argument 2 from ‘const std::shared_ptr<Component>’ to ‘std::shared_ptr<Component>&’
In file included from /usr/include/c++/4.8.2/algorithm:62:0,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/Board.h:6,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.h:3,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:1:
/usr/include/c++/4.8.2/bits/stl_algo.h:2266:34: error: no match for call to ‘(BRDBoard::BRDBoard(const BRDFile*)::__lambda1) (const std::shared_ptr<Component>&, std::shared_ptr<Component>&)’
    while (__comp(__pivot, *__last))
                                  ^
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:212:46: note: candidates are:
  sort(begin(components_), end(components_), [](shared_ptr<Component> &lhs, shared_ptr<Component> &rhs) {
                                              ^
In file included from /usr/include/c++/4.8.2/algorithm:62:0,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/Board.h:6,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.h:3,
                 from /builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:1:
/usr/include/c++/4.8.2/bits/stl_algo.h:2266:34: note: bool (*)(std::shared_ptr<Component>&, std::shared_ptr<Component>&) <conversion>
    while (__comp(__pivot, *__last))
                                  ^
/usr/include/c++/4.8.2/bits/stl_algo.h:2266:34: note:   candidate expects 3 arguments, 3 provided
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:212:102: note: BRDBoard::BRDBoard(const BRDFile*)::__lambda1
  sort(begin(components_), end(components_), [](shared_ptr<Component> &lhs, shared_ptr<Component> &rhs) {
                                                                                                      ^
/builddir/build/BUILD/OpenBoardView-R7.3/src/openboardview/BRDBoard.cpp:212:102: note:   no known conversion for argument 1 from ‘const std::shared_ptr<Component>’ to ‘std::shared_ptr<Component>&’

Hint: Try to edit PROJECT_VERSION bellow to make it apply to other versions.

---

--- openboardview/src/glad/CMakeLists.txt.orig	2016-10-14 23:37:00.287093058 +0200
+++ openboardview/src/glad/CMakeLists.txt	2016-10-14 23:37:44.962803034 +0200
@@ -1,4 +1,4 @@
-cmake_minimum_required(VERSION 3.0)
+cmake_minimum_required(VERSION 2.8)
 project(GLAD)
 
 set(GLAD_DIR "${PROJECT_SOURCE_DIR}")

--- openboardview/CMakeLists.txt.orig	2016-10-14 23:59:45.412222353 +0200
+++ openboardview/CMakeLists.txt	2016-10-15 00:01:43.206456420 +0200
@@ -1,4 +1,6 @@
-cmake_policy(SET CMP0048 OLD)
+if(POLICY CMP0048)
+	cmake_policy(SET CMP0048 OLD)
+endif(POLICY CMP0048)
 set(PROJECT_VERSION "R7.3")
 project(OpenBoardView)
 cmake_minimum_required(VERSION 2.8.9)
@@ -37,8 +39,8 @@
 
 # GCC version check
 if(CMAKE_CXX_COMPILER_ID STREQUAL "GNU")
-	if(CMAKE_CXX_COMPILER_VERSION VERSION_LESS "4.9")
-		message(FATAL_ERROR "GCC version needs to be >= 4.9")
+	if(CMAKE_CXX_COMPILER_VERSION VERSION_LESS "4.8")
+		message(FATAL_ERROR "GCC version needs to be >= 4.8")
 	endif()
 endif()
 
--- a/src/openboardview/FileFormats/BRDFile.h	2017-07-18 23:16:00.257871547 +0200
+++ b/src/openboardview/FileFormats/BRDFile.h	2017-07-18 23:18:00.961233575 +0200
@@ -88,7 +88,7 @@
 	static bool verifyFormat(std::vector<char> &buf);
 
   private:
-	static constexpr std::array<uint8_t, 4> signature = {0x23, 0xe2, 0x63, 0x28};
+	static constexpr std::array<uint8_t, 4> signature = {{0x23, 0xe2, 0x63, 0x28}};
 };
 
 char **stringfile(char *buffer);
 ```

piernov, 2017-07-19:
Fixed. If you need support for a CMake version lower than 3.0, you should ask on the glad project ( https://github.com/Dav1dde/glad ) if it's possible to lower the requirement.

closed 2017-07-19 by piernov

srakitnican, 2017-07-20:
Thank you very much!

For glad, opened here: https://github.com/Dav1dde/glad/issues/99

commit 381cc682 (2017-07-19): Merge pull request #78 from piernov/fix/gcc-4.8

Fix build with GCC 4.8 (closes #77)
commit b05b4166 (2017-07-19): Fix build with GCC 4.8 (closes #77)