OpenBoardView/OpenBoardView #83: Consider removing build.sh

MattSturgeon, 2017-07-20:
`build.sh` is a wrapper around the CMake build process intended to simplify the build process for new users.

In my opinion this makes little sense, CMake's entire purpose is to abstract the build process and allow users to easily set build options using CLI flags, a curses interface or even a GUI. `build.sh` on the other hand is another unnecessary layer of abstraction, another script to maintain, another potential point of failure.

If it greatly simplified the build process, it may be useful, however I would suggest that `cd build && cmake .. && make package` is simple enough to document in the README and provides the user with more control.

Currently the `build.sh` is relatively complex, maybe even more so than the CMake config, so it's worth asking is it needed.

If there is anything being configured within `build.sh` I would argue it should be done instead within the CMake scripts.

(Apologies to whoever wrote it if I sounded harsh!)

piernov, 2017-07-21:
`build.sh` doesn't configure anything. Everything is done through CMake.

Thus you can use you good old `mkdir build/; cd build; cmake ..; make -j? install; cd ..`, for which you have to determine the number parallel compilation process manually. And if you want to make a debug build, add `-DCMAKE_BUILD_TYPE=DEBUG` to your `cmake` command. And if you want to cross-compile for Windows, add `-DCMAKE_TOOLCHAIN_FILE=../Toolchain-mingw64.cmake`. And if you want to build everything anew, enter `rm -rf build/` before. And if you want to strip the binary, do `make install/strip` instead. Well, you see the idea.

It is indeed only a wrapper, but it makes my life way easier as a developper, and I think it would be the same for the end users and packagers. I personally don't like to dig through the documentation just to try some piece of software, and finding if this particular software is built using `maven`, `gradle`, `autotools`, `waf`, `scons`, `cmake`… is quite bothersome, and each of them might have a different build process. Sometimes you have a bare `Makefile` or just a source code file. Sometimes you have to add mandatory parameters, build in the same directory or in another directory… A simple `build.sh` script does all of this for you.

inflex, 2017-07-21:
For now I'd like to see the build.sh stay; it does provide a simple single-command front end and even if it's rendered moot in the future it can just be a wrapper to the "cd build && cmake && make package" sequence.

From personal experience if uninitiated to the project, it's nice to see a build.sh (or similar) single script to run rather than diving in to subdirectories.

MattSturgeon, 2017-07-21:
Closing since I'm heavily outvoted, plus good arguments.

closed 2017-07-21 by MattSturgeon