OpenBoardView/OpenBoardView #85: Drop DirectX 9 support

piernov, 2017-07-21:
We still have `main_dx9.cpp` and `imgui_impl_dx9.cpp` lying around, even though they are not used anymore for the official Windows build (since we cross-compile the OpenGL stuff). Should we delete them from the repository since they can be considered as dead code, or does anyone see any use for them?

MattSturgeon, 2017-07-21:
+1 for deleting any and all dead code

+1 to cleanup in general :)

piernov, 2017-07-22:
This will also remove support for the standalone VS project (which is broken anyway).

Support for building under Windows with MSVC is indeed broken too. Apart from small fixes needed here and there, the `lookup_file_insensitive()` function needs to be ported to win32 API as well as the use of `access()` in `main_opengl.cpp`.

The integration of CMake in VS2017 is quite interesting and seems to be working well. In our case we still have to install Python and Git, as well as manually downloading the SDL2 devel lib to the `src/SDL2` directory. Also it seems like SDL2 isn't linked statically, I guess there is a parameter for the MSVC compiler that needs to be added.

MattSturgeon, 2017-07-22:
IMO the static VS should go anyway too: VS projects should be generated by
CMake the same way it generates Makefiles on unix.

commit bcef96c2 (2017-07-23): Drop Dx9 support and standalone VS project (closes #85) (#87)
closed 2017-07-23 by piernov