OpenBoardView/OpenBoardView #90: Assertion Failed

portskea, 2017-07-31:
[image]

Hi!
Thank You for Your program. Im having an issue.

When trying to open a file with extension .fz the error shows up "Assertion Failed", expression content != nullptr

FZFile.cpp, line 234

Thank You

MattSturgeon, 2017-07-31:
Can you upload a copy of the file you were trying to use?

If you put it in a zip archive you should be able to [drag and drop](https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/) it into a comment here.

piernov, 2017-07-31:
#74

MattSturgeon, 2017-07-31:
Ah, forgot about the encryption.

I'll open a seperate issue to add a warning if the fz key is empty and close this as a duplicate.

closed 2017-07-31 by MattSturgeon

portskea, 2017-08-01:
File attach. Thank You!

binilj04, 2022-07-07:
For those coming here for FZ key null error:

File -> Program Preference

https://gist.github.com/binilj04/db5539574027949c1c670fba8c837b24

[source link here](https://belitong.wordpress.com/tag/openboardview/)