OpenBoardView/OpenBoardView #91: Warn the user if they open a FZ file without a key defined

MattSturgeon, 2017-07-31:
#74 (duplicated by #90) was caused by FZFile failing at [`ENSURE(content != nullptr)`](https://github.com/OpenBoardView/OpenBoardView/blob/508e4d0f0b2b26c839387a1a7b72acaa0a498337/src/openboardview/FileFormats/FZFile.cpp#L234) after decryption fails due to the key being invalid.

Instead, we should check if a key exists _before_ creating a FZFile instance, and perhaps also pass the key in as an argument to FZFile. Perhaps something like:

```cpp
if (check_fileext(filename, ".fz")) {
    if (CheckFZKey(FZKey)) {
        file = new FZFile(buffer, FZKey);
    }
    else {
        print_cool_imgui_warning("Warn users about FZKey!!");
    }
}
```

MattSturgeon, 2017-08-02:
We already handle `file.valid` being false, and if it is a fz file we even show the user the FZKey. Perhaps all we should do is set `valid = 0` if FZFile somehow fails. @piernov thoughts?

wtangerine, 2017-10-23:
It might be a good idea to present an interactive prompt when the key is wrong or missing so the user could enter the key right away. That's what 7z or winrar do in similar cases.

commit 72acc29f (2022-03-20): Validate parity of FZ key or report error (closes #91)
closed 2022-03-20 by piernov