OpenBoardView/OpenBoardView #93: Screensaver and Sleep disabled when OBV is open on OSX

tiagobaracho, 2017-08-03:
IF OBV is open on OSX ( in my case sierra) it makes screensaver never came up nor the sleep / monitor off.... I have to close OBV to screensaver and sleep back to work.
Thanks guys .

inflex, 2017-08-03:
I better get my macbook setup with the build tools.

I'll be curious to see what's causing this. Might need to use something other than usleep() perhaps on the macOS to enable sleeping, or maybe it's the SDL polling (even SDL_waitevent() isn't any good, as it seems to be just a wrapper around the same polling, as opposed to true event-driven ).

MattSturgeon, 2017-08-03:
Maybe OSX is treating us as a game/video due to the opengl rendering and not wanting to interrupt the user playing/watching their media?

inflex, 2017-08-03:
That's a plausible option.  I'll set it up and see what's happening with the main loop, and I'll dig around to see if there's any talk about SDL/OpenGL keeping macOS awake.

piernov, 2017-08-03:
```
Hikari:~ piernov$ pmset -g assertions
2017-08-03 16:52:59 +0200 
Assertion status system-wide:
   BackgroundTask                 1
   ApplePushServiceTask           0
   UserIsActive                   1
   PreventUserIdleDisplaySleep    1
   PreventSystemSleep             0
   ExternalMedia                  1
   PreventUserIdleSystemSleep     1
   NetworkClientActive            0
Listed by owning process:
   pid 217(nsurlsessiond): [0x0000012b000b80f0] 00:02:19 BackgroundTask named: "NSURLSessionTask F61AE2A7-8250-440D-8689-93C6BD00F213" 
   pid 217(nsurlsessiond): [0x000000d7000b80e3] 00:03:43 BackgroundTask named: "NSURLSessionTask 37E1827B-DCE7-4EF3-BE49-04812D3489F6" 
   pid 217(nsurlsessiond): [0x000000cf000b80dd] 00:03:51 BackgroundTask named: "NSURLSessionTask 7BA1B1FC-74A0-4D43-9901-EC76A0FAD687" 
   pid 566(openboardview): [0x00000099000580a1] 00:04:45 PreventUserIdleDisplaySleep named: "openboardview using SDL_DisableScreenSaver" 
   pid 44(powerd): [0x0000002a00088000] 00:06:37 ExternalMedia named: "com.apple.powermanagement.externalmediamounted" 
   pid 97(hidd): [0x0000006c0009803b] 00:05:30 UserIsActive named: "com.apple.iohideventsystem.queue.tickle.4294968320.3" 
        Timeout will fire in 597 secs Action=TimeoutActionRelease
   pid 561(softwareupdated): [0x0000016b000180f9] 00:01:14 PreventUserIdleSystemSleep named: "Updating Software (com.apple.SoftwareUpdate.Session)" 
        Details: Downloading 
   pid 561(softwareupdated): [0x0000010d000180e8] 00:02:49 PreventUserIdleSystemSleep named: "Updating Software (com.apple.SoftwareUpdate.Session)" 
        Details: Downloading iTunesX-12.6.2
   pid 473(storeassetd): [0x00000169000b80f8] 00:01:16 BackgroundTask named: "com.apple.storeagent.UpdateStaging" 
   pid 473(storeassetd): [0x0000010b000b80e7] 00:02:51 BackgroundTask named: "com.apple.storeagent.UpdateStaging" 
Kernel Assertions: 0x104=USB,MAGICWAKE
   id=502  level=255 0x4=USB mod=01/01/1970 01:00 description=com.apple.usb.externaldevice.14300000 owner=Generic USB Hub
   id=504  level=255 0x4=USB mod=01/01/1970 01:00 description=com.apple.usb.externaldevice.14320000 owner=USB OPTICAL MOUSE 
   id=505  level=255 0x100=MAGICWAKE mod=01/01/1970 01:00 description=en0 owner=en0
   id=506  level=255 0x4=USB mod=01/01/1970 01:00 description=com.apple.usb.externaldevice.14310000 owner=USB Wireless HID Receiver
Idle sleep preventers: IODisplayWrangler
```

`SDL_DisableScreenSaver` might be called by default. Maybe we need to explicitly call `SDL_EnableScreenSaver`.

piernov, 2017-08-03:
Fucked up my hackintosh again just before I could try that… Well, here is the possible fix: piernov/OpenBoardView@6d2637896916dcf8c9d39b843f6c8c5355edbe42 
Travis will be uploading the builds shortly here: https://github.com/piernov/OpenBoardView/releases/tag/N15 , @tiagobaracho please try that on both Windows and macOS.

inflex, 2017-08-03:
I'll try it on my macbook as well, thanks @piernov