OpenBoardView/OpenBoardView #110: Teradyne GenCAD format

martonmiklos, 2017-09-27:
I have met this format at several companies in the industry mainly used for board level debugging.
The format is text based and has accessible documentation:
http://electronix.ru/forum/index.php?act=Attach&type=post&id=15091

I am not aware of any opensource/free tools which is capable opening this format. From the properitry tools I only know the Teradyne D2B which supports it: http://www.teradyne.com/products/production-board-test/design-to-build-d2b-software

At some point I might come up with a PR.

martonmiklos, 2018-11-28:
Side note to myself (and might be others planning to add support for text based formats):
https://github.com/orangeduck/mpc

This library could ease the parser creation pretty much. I hope I will get some time to implement this.

martonmiklos, 2019-08-15:
In the case if anyone would miss this I am cooking it right now here:
https://github.com/martonmiklos/OpenBoardView/tree/gencad

Basically rolls, just needs some further optimisations. (I will throw out the unused stuff from the parsing.)

bunker-bunk567, 2021-09-15:
> In the case if anyone would miss this I am cooking it right now here:
> https://github.com/martonmiklos/OpenBoardView/tree/gencad
> 
> Basically rolls, just needs some further optimisations. (I will throw out the unused stuff from the parsing.)

why is this not being merged into main?

closed 2022-06-14 by piernov