OpenBoardView/OpenBoardView #112: C++ Runtime error when open fz file

Mark-Joy, 2017-10-16:
Hello I got below C++ Runtime error when open fz file. Could you help to solve the issue

[image]

Board file and obv.conf file: 
[Schematics.zip](https://github.com/OpenBoardView/OpenBoardView/files/1388353/Schematics.zip)

Thank you!

piernov, 2017-10-16:
I will take a look at it. In the meantime, please try with the files in the `.zip` from #10 .

Mark-Joy, 2017-10-18:
Thank you.
I accidentally found one program can also open this fz file: http://boardviewer.net/

piernov, 2017-10-19:
There was some special section with drawings inside this file. They are now ignored rather than parsed as parts (which caused the crash).
I don't know what they are for.

closed 2017-10-19 by piernov

wtangerine, 2017-10-20:
Thanks for fixing (waiting patiently for a Windows build with the fix).

It would be nice if there was a way to extract these drawings.

piernov, 2017-10-20:
I'm gonna do a release as soon as I fix some stuff about the OpenGL version choice. For now I'll start a build on my fork: https://github.com/piernov/OpenBoardView/releases .

Yep, sadly I don't have the time to implement that right now. :(

Mark-Joy, 2017-10-20:
Thank you for the fix. But the board is not showing correctly, it is missing most of the components. Left side is from openboardview, right size is from boardview.net

[image]

wtangerine, 2017-10-21:
Mark-Joy, you are looking at the bottom side of the board with openboardview, and at the top side with the other app.

Tested here - the pre-release version of openboardview is working correctly with your file and seems to be displaying all the components.

commit 7f366f34 (2017-10-19): Ignore sections not implemented on FZ file (closes #112)
commit e18aceaf (2017-10-19): Merge pull request #113 from piernov/fix/fz-not-implemented-sections

Ignore sections not implemented on FZ file (closes #112)