OpenBoardView/OpenBoardView #114: MinGW Runtime Assertion when opening Asus fz files

wtangerine, 2017-10-21:
Tried to open 2 fz files from asus and both produce the same error.
Both files open normally with the boardviewer v1.2 app.

[image]

[Asus boardview.zip](https://github.com/OpenBoardView/OpenBoardView/files/1404391/Asus.boardview.zip)

piernov, 2017-10-21:
Works fine here. I guess you didn't add the FZ key to OpenBoardView's configuration.

wtangerine, 2017-10-22:
Ahhh, now I get it. Found this discussion which explains it:
https://www.badcaps.net/forum/showthread.php?p=687140

Will go try to find the key (or can someone PM me with one?).

In the meanwhile, I think it would be a good idea to handle wrong/missing key more gracefully rather than basically crash, and present some meaningful info to the user.

piernov, 2017-10-22:
See #91 #90 #74

wtangerine, 2017-10-23:
Yeah, this should be closed as duplicate. Thanks!

closed 2019-12-28 by piernov