OpenBoardView/OpenBoardView #118: Use HTTPS for the website

C0rn3j, 2017-12-13:
Hiya, I see that you're using HTTP only (because HTTPS at the moment has a bad TLS/SSL cert), it would be good if the site was HTTPS only with HTTP redirecting to HTTPS, as the site links to binaries and such.

Let's Encrypt gives free certs, more arguments for HTTPS can be found here: https://doesmysiteneedhttps.com/

piernov, 2017-12-13:
@inflex is hosting the website. He already uses Let's Encrypt, but it seems like he forgot to add openboardview.org domain to his certificate. The certificate itself is valid, but for other domains.

inflex, 2017-12-13:
OBV domain added; not sure how long that'll take to propagate.

piernov, 2017-12-13:
@inflex certificate is OK. Web server config seems broken though, https://openboardview.org gives the same web page as https://ctpc.biz/ .

inflex, 2017-12-13:
@piernov should be resolved now.  Not sure why but the certbot didn't generate the LE/SSL config file for Apache, so I just cloned another one and modified.

C0rn3j, 2017-12-13:
Cool!

Now to just set redirect from :80 to :443.

As a bonus you can set the HSTS header and possibly join the HSTS preload list ^^ 
https://hstspreload.org/

C0rn3j, 2018-04-08:
I do, as per my last comment. Site is still available over HTTP, which it shouldn't. 
Don't think opening a new issue just for that makes sense.