OpenBoardView/OpenBoardView #120: Acer .brd (Allegro) file does not open

wtangerine, 2018-01-27:
[Quanta ZQX DA0ZQXMB8E0 REV 0.1 - BOARDVIEW.brd.zip](https://github.com/OpenBoardView/OpenBoardView/files/1670181/Quanta.ZQX.DA0ZQXMB8E0.REV.0.1.-.BOARDVIEW.brd.zip)

The message only says "There was an error opening the file ..."

The only software that seems to currently support the Allegro format is Allegro Free Physical Viewer. But that software is a nightmare to use.

File attached.

wtangerine, 2018-01-27:
Just saw issue #62 - probably this is a duplicate? I searched by the Allegro keywoard (which AFAIK is the name of the format) and didn't find anything, so I opened this bug.

inflex, 2018-01-27:
@piernov any thoughts?

inflex, 2018-01-27:
Looks to be another clear-data binary format.  Got to love all these .brd files with different formats.

piernov, 2018-01-27:
Yep, no one is working on supporting new formats so sadly this won't be supported anytime soon.

RavetcoFX, 2019-02-23:
Where can I start in the codebase to get something like this working?

piernov, 2019-02-24:
First thing would be to understand the format (which is the difficult part, especially with a huge binary file like that), then implement a decoder, taking inspiration from an existing one in `src/openboardview/FileFormats` and add call the decoder in the `LoadFile` function in `src/openboardview/BoardView.cpp`

piernov, 2019-07-26:
Closing as duplicate of #62

closed 2019-07-26 by piernov