OpenBoardView/OpenBoardView #124: Error opening a Board View

DinBan, 2018-04-17:
Hello and thank you for the openboardview.
But it can't open many of BoardView *.pcb
It just says "There was an error opening file...
OpenBoardView R534, Mac OS High Sierra 10.13.4
PCB file attached.
Any solution?
Thank you in advance!
[image]
"
[image]

[iPhone6 -4.7.pcb.zip](https://github.com/OpenBoardView/OpenBoardView/files/1920443/iPhone6.-4.7.pcb.zip)

inflex, 2018-04-17:
We have not been able to reverse engineer the format sufficiently.  Thus far no one has been able to furnish us with a specification either.

DinBan, 2018-04-17:
#60 for Windows only....

closed 2018-05-13 by piernov