OpenBoardView/OpenBoardView #126: BRD file can't open with open board viewer

Tom12984, 2018-04-30:
Hi Sir
 
Yesterday, that I download free version "open board viewer #7.3" but its seem gone "the message show can't open BRD file". Had you chance to help to check that ! thank in abvance

BTW:How can we send file for you check !
BR
Tom

[77G MMW automotive radar-PCBF-V01-20180115-1520.zip](https://github.com/OpenBoardView/OpenBoardView/files/1959369/77G.MMW.automotive.radar-PCBF-V01-20180115-1520.zip)

pvtinflex, 2018-05-28:
This is another Cadence Allegro board

inflex, 2018-05-28:
This is a Cadence Allegro format.  Not currently supported.

anshadedavana, 2018-07-07:
I too have a BRD file which cant be opened with OBV but can be viewed with IDEAL ELECTRONICS BOARDIEW and HE BOARDVIEW apps. Is this also Allegro format ?. 
[BoardView.zip](https://github.com/OpenBoardView/OpenBoardView/files/2172518/BoardView.zip)

inflex, 2018-07-07:
No, this one is something new / different again.  At least this one is plain-text and probably can be decoded with a day or two of work.

No shortage of file formats using the .BRD extension.

inflex, 2018-07-07:
@anshadedavana can you create a new thread with your file,  call it "New Landrex-BRD format" or similar.

anshadedavana, 2018-07-07:
Many thanks for consider adding this to the OBV Paul. I have many Acer
laptop boardiew files which are in this same format.  If it helps, Remax's
BoardViewer 1.2  app can open this format without issues. However i prefer
OBV as its the best BoardViewer i have used so far and probably in the
future too. Please accept my gratitude for creating this amazing peace of
software. I am just a starter in this field so dont make much money now.
However will definitely consider buying the paid version when i get
financially sound.

leiangelo05, 2018-08-14:
hi, same i get the same error
[Oxford MB-A01-MODIFY01.zip](https://github.com/OpenBoardView/OpenBoardView/files/2286164/Oxford.MB-A01-MODIFY01.zip)

johnfordjr, 2018-10-18:
I'm lost.... Been using OpenBoardView with no problems (still worked perfectly March of 2018) 
Then all of a sudden it won't open .BRD or .BV files (same files that were working previously).
 What am i missing? Who knows? I would be happy to pay for the software but alas I can't use the software anymore. Now I'm feeling sad and stupid. Is there a newer version than Build R534 R7.3??

piernov, 2019-07-26:
Closing in favor of #134 and #62

closed 2019-07-26 by piernov